RobotLocomotion/drake #18089: CI failure: clang++ segfault on mac

ggould-tri, 2022-10-12:
In https://drake-jenkins.csail.mit.edu/view/Production/job/mac-big-sur-unprovisioned-clang-bazel-nightly-snopt-mosek-packaging/181/ the invocation of `wrapped_clang_pp` to compile `multibody/plant/tamsi_solver.cc` failed with a segfault.
```
[3:48:42 AM]  [4,829 / 6,216] Compiling multibody/plant/tamsi_solver.cc; 4s darwin-sandbox ... (3 actions running)
[3:48:51 AM]  ERROR: /Users/bigsur/workspace/mac-big-sur-unprovisioned-clang-bazel-nightly-snopt-mosek-packaging/src/multibody/plant/BUILD.bazel:67:17: Compiling multibody/plant/tamsi_solver.cc failed: (Aborted): wrapped_clang_pp failed: error executing command external/local_config_cc/wrapped_clang_pp '-D_FORTIFY_SOURCE=1' -fstack-protector -fcolor-diagnostics -Wall -Wthread-safety -Wself-assign -fno-omit-frame-pointer -g0 -O2 -DNDEBUG ... (remaining 107 arguments skipped)
[3:48:51 AM]  
[3:48:51 AM]  Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
[3:48:51 AM]  clang++: error: unable to execute command: Segmentation fault: 11
[3:48:51 AM]  clang++: error: clang frontend command failed due to signal (use -v to see invocation)
[3:48:51 AM]  Apple clang version 12.0.5 (clang-1205.0.22.9)
[3:48:51 AM]  Target: x86_64-apple-macosx11.3
[3:48:51 AM]  Thread model: posix
[3:48:51 AM]  InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
[3:48:51 AM]  clang++: note: diagnostic msg: 
[3:48:51 AM]  ********************
[3:48:51 AM]  
[3:48:51 AM]  PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
[3:48:51 AM]  Preprocessed source(s) and associated run script(s) are located at:
[3:48:51 AM]  clang++: note: diagnostic msg: /var/folders/gd/65m3ns0s6b52wk780qn5lp0r0000gn/T/tamsi_solver-1ca0ae.cpp
[3:48:51 AM]  clang++: note: diagnostic msg: /var/folders/gd/65m3ns0s6b52wk780qn5lp0r0000gn/T/tamsi_solver-1ca0ae.sh
[3:48:51 AM]  clang++: note: diagnostic msg: Crash backtrace is located in
[3:48:51 AM]  clang++: note: diagnostic msg: /Users/bigsur/Library/Logs/DiagnosticReports/clang++_<YYYY-MM-DD-HHMMSS>_<hostname>.crash
[3:48:51 AM]  clang++: note: diagnostic msg: (choose the .crash file that corresponds to your crash)
[3:48:51 AM]  clang++: note: diagnostic msg: 
[3:48:51 AM]  
[3:48:51 AM]  ********************
[3:48:51 AM]  Error in child process '/usr/bin/xcrun'. 254
```
Opening and immediately closing this bug as probably a one-off (what's the likelihood we'd find a reproducible ICE?).  If this happens again, re-open it.

ggould-tri, 2023-01-10:
A nearly identical compiler segfault today, this time in the mosek test.
 * https://drake-jenkins.csail.mit.edu/view/Production/job/mac-x86-monterey-clang-bazel-nightly-everything-address-sanitizer/62/
```
[4:01:28 AM]  ERROR: /Users/monterey/workspace/mac-x86-monterey-clang-bazel-nightly-everything-address-sanitizer/src/solvers/BUILD.bazel:1457:29: Compiling solvers/test/mosek_solver_internal_test.cc failed: (Aborted): wrapped_clang_pp failed: error executing command external/local_config_cc/wrapped_clang_pp '-D_FORTIFY_SOURCE=1' -fstack-protector -fcolor-diagnostics -Wall -Wthread-safety -Wself-assign -fno-omit-frame-pointer -g '-std=c++11' 'DEBUG_PREFIX_MAP_PWD=.' ... (remaining 176 arguments skipped)
[4:01:28 AM]  
[4:01:28 AM]  Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
[4:01:28 AM]  clang++: error: unable to execute command: Segmentation fault: 11
[4:01:28 AM]  clang++: error: clang frontend command failed due to signal (use -v to see invocation)
[4:01:28 AM]  Apple clang version 14.0.0 (clang-1400.0.29.102)
[4:01:28 AM]  Target: x86_64-apple-macosx12.3
[4:01:28 AM]  Thread model: posix
[4:01:28 AM]  InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
[4:01:28 AM]  clang++: note: diagnostic msg: 
[4:01:28 AM]  ********************
[4:01:28 AM]  
[4:01:28 AM]  PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
[4:01:28 AM]  Preprocessed source(s) and associated run script(s) are located at:
[4:01:28 AM]  clang++: note: diagnostic msg: /var/folders/c8/66b8jsn95tl1b7l4mx6bn04r0000gp/T/mosek_solver_internal_test-d5c907.cpp
[4:01:28 AM]  clang++: note: diagnostic msg: /var/folders/c8/66b8jsn95tl1b7l4mx6bn04r0000gp/T/mosek_solver_internal_test-d5c907.sh
[4:01:28 AM]  clang++: note: diagnostic msg: Crash backtrace is located in
[4:01:28 AM]  clang++: note: diagnostic msg: /Users/monterey/Library/Logs/DiagnosticReports/clang++_<YYYY-MM-DD-HHMMSS>_<hostname>.crash
[4:01:28 AM]  clang++: note: diagnostic msg: (choose the .crash file that corresponds to your crash)
[4:01:28 AM]  clang++: note: diagnostic msg: 
[4:01:28 AM]  
[4:01:28 AM]  ********************
[4:01:28 AM]  Error in child process '/usr/bin/xcrun'. 254
```

ggould-tri, 2023-01-10:
Reopening for triage, since this has now recurred.

BetsyMcPhail, 2023-01-16:
Another instance: https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-x86-monterey-unprovisioned-clang-bazel-nightly-release/66/

```
[3:27:32 AM]  ERROR: /Users/monterey/workspace/mac-x86-monterey-unprovisioned-clang-bazel-nightly-release/src/common/symbolic/BUILD.bazel:230:20: Compiling common/symbolic/test/polynomial_basis_test.cc failed: (Aborted): wrapped_clang_pp failed: error executing command external/local_config_cc/wrapped_clang_pp '-D_FORTIFY_SOURCE=1' -fstack-protector -fcolor-diagnostics -Wall -Wthread-safety -Wself-assign -fno-omit-frame-pointer -g0 -O2 -DNDEBUG ... (remaining 145 arguments skipped)
[3:27:32 AM]  
[3:27:32 AM]  Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
[3:27:32 AM]  clang++: error: unable to execute command: Segmentation fault: 11
[3:27:32 AM]  clang++: error: clang frontend command failed due to signal (use -v to see invocation)
[3:27:32 AM]  Apple clang version 14.0.0 (clang-1400.0.29.102)
[3:27:32 AM]  Target: x86_64-apple-macosx12.3
[3:27:32 AM]  Thread model: posix
[3:27:32 AM]  InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
[3:27:32 AM]  clang++: note: diagnostic msg: 
[3:27:32 AM]  ********************
[3:27:32 AM]  
[3:27:32 AM]  PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
[3:27:32 AM]  Preprocessed source(s) and associated run script(s) are located at:
[3:27:32 AM]  clang++: note: diagnostic msg: /var/folders/c8/66b8jsn95tl1b7l4mx6bn04r0000gp/T/polynomial_basis_test-09a9f9.cpp
[3:27:32 AM]  clang++: note: diagnostic msg: /var/folders/c8/66b8jsn95tl1b7l4mx6bn04r0000gp/T/polynomial_basis_test-09a9f9.sh
[3:27:32 AM]  clang++: note: diagnostic msg: Crash backtrace is located in
[3:27:32 AM]  clang++: note: diagnostic msg: /Users/monterey/Library/Logs/DiagnosticReports/clang++_<YYYY-MM-DD-HHMMSS>_<hostname>.crash
[3:27:32 AM]  clang++: note: diagnostic msg: (choose the .crash file that corresponds to your crash)
[3:27:32 AM]  clang++: note: diagnostic msg: 
[3:27:32 AM]  
[3:27:32 AM]  ********************
[3:27:32 AM]  Error in child process '/usr/bin/xcrun'. 254
[3:28:41 AM]  ERROR: /Users/monterey/workspace/mac-x86-monterey-unprovisioned-clang-bazel-nightly-release/src/multibody/optimization/BUILD.bazel:282:20: Compiling multibody/optimization/test/quaternion_integration_constraint_test.cc failed: (Aborted): wrapped_clang_pp failed: error executing command external/local_config_cc/wrapped_clang_pp '-D_FORTIFY_SOURCE=1' -fstack-protector -fcolor-diagnostics -Wall -Wthread-safety -Wself-assign -fno-omit-frame-pointer -g0 -O2 -DNDEBUG ... (remaining 416 arguments skipped)
[3:28:41 AM]  
[3:28:41 AM]  Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
[3:28:41 AM]  clang++: error: unable to execute command: Segmentation fault: 11
[3:28:41 AM]  clang++: error: clang frontend command failed due to signal (use -v to see invocation)
[3:28:41 AM]  Apple clang version 14.0.0 (clang-1400.0.29.102)
[3:28:41 AM]  Target: x86_64-apple-macosx12.3
[3:28:41 AM]  Thread model: posix
[3:28:41 AM]  InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
[3:28:41 AM]  clang++: note: diagnostic msg: 
[3:28:41 AM]  ********************
[3:28:41 AM]  
[3:28:41 AM]  PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
[3:28:41 AM]  Preprocessed source(s) and associated run script(s) are located at:
[3:28:41 AM]  clang++: note: diagnostic msg: /var/folders/c8/66b8jsn95tl1b7l4mx6bn04r0000gp/T/quaternion_integration_constraint_test-d5da2b.cpp
[3:28:41 AM]  clang++: note: diagnostic msg: /var/folders/c8/66b8jsn95tl1b7l4mx6bn04r0000gp/T/quaternion_integration_constraint_test-d5da2b.sh
[3:28:41 AM]  clang++: note: diagnostic msg: Crash backtrace is located in
[3:28:41 AM]  clang++: note: diagnostic msg: /Users/monterey/Library/Logs/DiagnosticReports/clang++_<YYYY-MM-DD-HHMMSS>_<hostname>.crash
[3:28:41 AM]  clang++: note: diagnostic msg: (choose the .crash file that corresponds to your crash)
[3:28:41 AM]  clang++: note: diagnostic msg: 
[3:28:41 AM]  
[3:28:41 AM]  ********************
[3:28:41 AM]  Error in child process '/usr/bin/xcrun'. 254
[3:28:41 AM]  ERROR: /Users/monterey/workspace/mac-x86-monterey-unprovisioned-clang-bazel-nightly-release/src/common/symbolic/expression/BUILD.bazel:126:20: Compiling common/symbolic/expression/test/expression_matrix_test.cc failed: (Aborted): wrapped_clang_pp failed: error executing command external/local_config_cc/wrapped_clang_pp '-D_FORTIFY_SOURCE=1' -fstack-protector -fcolor-diagnostics -Wall -Wthread-safety -Wself-assign -fno-omit-frame-pointer -g0 -O2 -DNDEBUG ... (remaining 145 arguments skipped)
[3:28:41 AM]  
[3:28:41 AM]  Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
[3:28:41 AM]  clang++: error: unable to execute command: Segmentation fault: 11
[3:28:41 AM]  clang++: error: clang frontend command failed due to signal (use -v to see invocation)
[3:28:41 AM]  Apple clang version 14.0.0 (clang-1400.0.29.102)
[3:28:41 AM]  Target: x86_64-apple-macosx12.3
[3:28:41 AM]  Thread model: posix
[3:28:41 AM]  InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
[3:28:41 AM]  clang++: note: diagnostic msg: 
[3:28:41 AM]  ********************
[3:28:41 AM]  
[... cut ...]
[3:28:42 AM]  ==================== Test output for //multibody/optimization:py/static_equilibrium_problem_drakelint:
[3:28:42 AM]  ================================================================================
[3:28:42 AM]  FAIL: //multibody/optimization:py/static_equilibrium_problem_drakelint (see /Users/monterey/workspace/mac-x86-monterey-unprovisioned-clang-bazel-nightly-release/_bazel_monterey/ebfe24e3348b8315117ff571aa22810d/execroot/drake/bazel-out/darwin-opt/testlogs/multibody/optimization/py/static_equilibrium_problem_drakelint/test.log)
```

SeanCurtis-TRI, 2023-02-28:
Another instance last night:
https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-monterey-unprovisioned-clang-bazel-nightly-release/145/

jwnimmer-tri, 2023-09-23:
This issue is too rare and intermittent merit an open ticket.

closed 2023-09-23 by jwnimmer-tri

RussTedrake, 2023-09-23:
fwiw -- i do think we have users that frequency fail to compile drake on their first try due to running out of memory.  (I presume something like that is happening here?)