RobotLocomotion/drake #18091: Remaining Jammy CI builds

jwnimmer-tri, 2022-10-12:
#16217 tracked adding Jammy support. In that issue, passing jobs were moved into production.  This issue covers the remainder.

We still have four Jammy jobs that are excluded from Production:

```
linux-jammy-clang-bazel-weekly-non-production-gurobi-thread-sanitizer
linux-jammy-clang-bazel-weekly-non-production-mosek-thread-sanitizer
linux-jammy-clang-bazel-weekly-non-production-snopt-thread-sanitizer
linux-jammy-clang-bazel-weekly-non-production-mosek-valgrind-memcheck
```

We should work to either enable them, or delete them as out-of-scope.

I suspect that #17562 is the only blocker for turning on the thread sanitizer builds.

I'm not sure if we've looked into the mosek-valgrind one.

BetsyMcPhail, 2022-11-21:
When the DockerHub password was updated this summer, only the production jobs were updated. https://github.com/RobotLocomotion/drake-jenkins-jobs/pull/58 also updates the the non-production jobs.

jwnimmer-tri, 2023-01-05:
See #18098 for a possible fix for TSan.

BetsyMcPhail, 2023-01-05:
There is one failing test on `linux-jammy-clang-bazel-weekly-non-production-mosek-valgrind-memcheck` - [//solvers:mosek_solver_internal_test](https://drake-cdash.csail.mit.edu/viewTest.php?onlyfailed&buildid=1659179). With the goal of moving Jammy jobs to production, @williamjallen is going to skip memcheck on that test.

BetsyMcPhail, 2023-01-06:
The `linux-jammy-clang-bazel-weekly-non-production-mosek-valgrind-memcheck` job has been moved to production.

@jwnimmer-tri are we ready to move the other jobs into production? (I'm updating the Jammy AMI  now for #18541)

jwnimmer-tri, 2023-01-06:
> @jwnimmer-tri are we ready to move the other jobs into production?

We should just run them in CI and see whether they have errors (after #18541 is in effect).

BetsyMcPhail, 2023-01-06:
Experimental jobs using updated Jammy AMI:

https://drake-jenkins.csail.mit.edu/view/Linux%20Jammy/job/linux-jammy-clang-bazel-experimental-gurobi-thread-sanitizer/2/
https://drake-jenkins.csail.mit.edu/view/Linux%20Jammy/job/linux-jammy-clang-bazel-experimental-mosek-thread-sanitizer/2/
https://drake-jenkins.csail.mit.edu/view/Linux%20Jammy/job/linux-jammy-clang-bazel-experimental-snopt-thread-sanitizer/2/

jwnimmer-tri, 2023-01-06:
Oops, I also just relaunched the "weeky non-production" builds at https://drake-jenkins.csail.mit.edu/view/ThreadSanitizer/.  I suppose the results will be identical.

jwnimmer-tri, 2023-01-06:
The experimental jobs all passed.  I think we're ready to promote jobs to production.

BetsyMcPhail, 2023-01-06:
@williamjallen can you please create a PR to move the remainder of the weekly non-production jobs to production?

closed 2023-01-10 by williamjallen