RobotLocomotion/drake #18097: [tutorials] Update Hydroelastic Contact User Guide

DamrongGuoy, 2022-10-13:
### Description
With the new development in the hydroelastic contact model, it is a good time to improve the [Hydroelastic Contact User Guide.](https://drake.mit.edu/doxygen_cxx/group__hydroelastic__user__guide.html)

- Make it clear how the three [ContactModel](https://drake.mit.edu/doxygen_cxx/namespacedrake_1_1multibody.html#a6f6e054758d34b83e86bea87d65dfe4e)s work by adding tables in the Appendix below.
- All shape specifications are supported for both rigid and compliant now. The last one is the compliant non-convex Mesh.
- Pictures of both rigid-compliant contact surfaces and compliant-compliant contact surfaces.
- Short descriptions of all examples in [drake/examples/hydroelastic/](https://github.com/RobotLocomotion/drake/tree/master/examples/hydroelastic), so users can study further.

### Appendix
We can use these tables to describe the three contact models.
[image]

DamrongGuoy, 2022-10-13:
CC: @rcory

jwnimmer-tri, 2022-10-13:
When you make the tables for real in the website docs, I advise against using single-letter "encryption" within each cell to explain the point.   We can probably afford to write out "Throws", "Hydroelastic", etc.

jwnimmer-tri, 2024-11-01:
@amcastro-tri @joemasterjohn @DamrongGuoy This ticket is a few years old now.  What (if any) missing text do we specifically still need in our guides and/or tutorials?