RobotLocomotion/drake #18103: MathematicalProgram stores the summed-up linear and quadratic costs

hongkai-dai, 2022-10-13:
Currently MathematicalProgram stores a vector of `LinearCost` and `QuadraticCost` objects, while all the downstream solvers need the summed up costs
```
0.5x'Qx + b'x + c
````
Besides, if we know the summed-up costs, it is easier to determine if the cost is convex or not.

So the proposal is to store the matrix `Q`, `b` and constant `c` in addition to the vector of `LinearCost` and `QuadraticCost` in MathematicalProgram.

The `Q` matrix and `b` vector is often sparse. I am thinking to store it in this way
```
std::unordered_map<int, std::unordered_map<int, double>> Q_;
std::unordered_map<int, double> b_;
```
where Q_[i][j] stores Q(i, j), b_[i] stores b(i). Storing both sparse matrices as `unordered_map` allows us to easily update the matrix coefficient, and allows flexibility to append new variables to the program (as compared to storing Eigen::VectorXd b_ which would require resizing the vector `b` every time we append new variables).

@jwnimmer-tri what do you think?

jwnimmer-tri, 2022-10-14:
If all (or at least most) of the solver back-ends want a unified Q and b, then I think it makes sense for the MathematicalProgram to provide that data easily.

As to the choice for storing the sparse `Q` and `b`, I would have expected to use `Eigen::Sparse` .  Is there some downside to using it?  From a quick skim, it seems like it would do some reallocation upon growth, but perhaps not as much as a full `MatrixXd`.

On the other hand, why do we need to compute `Q` and `v` incrementally?  It seems like it would create both of them from the list of costs on demand (at the very end, when the back-end asks us for them), so there would no resizing anyway.

Another option would be to collect a list of pending triplets incrementally, and then move those into `Q` and `b` anytime we need the full matrix.

Another option would be to have an accessor function `MP::GetCompositeQuadraticCostMatrix(Eigen::Sparse* Q)` with `Q` as an output argument.  Then the back-end could just call that function to get Q, no need to pre-compute it at all.

jwnimmer-tri, 2022-10-14:
Looking again, I don't think the MathematicalProgram can store `Q` ahead of time.  The user can call `QuadraticCost::UpdateCoefficients` while setting up the program, and the program will never know about it -- any pre-calculated composite Q will go out of sync with the actual costs.

I think what we should focus on is for the `Solve()` to have a _convenient_ way to fetch the composite `Q`.  So I think something like my final suggestion above is a good direction to explore.

hongkai-dai, 2022-10-14:
> Looking again, I don't think the MathematicalProgram can store Q ahead of time. The user can call QuadraticCost::UpdateCoefficients while setting up the program, and the program will never know about it -- any pre-calculated composite Q will go out of sync with the actual costs.

Good point, I forgot UpdateCoefficients.

> Another option would be to have an accessor function MP::GetCompositeQuadraticCostMatrix(Eigen::Sparse* Q) with Q as an output argument. Then the back-end could just call that function to get Q, no need to pre-compute it at all.

Currently we have AggregateQuadraticCosts https://github.com/RobotLocomotion/drake/blob/09b32778c53547f6413945224ee7a5874cc1d64c/solvers/aggregate_costs_constraints.h#L39-L45 that will compute `Q` from a list of QuadraticCosts. I don't think we have to move this function to be a method of MathematicalProgram().

So I think we could still add `MathematicalProgram::IsConvex()` method, which computes `Q` matrix and then determines if this Q is positive semidefinite or not.

hongkai-dai, 2022-10-17:
I think one solution is like this: when we determine if the total cost is convex or not, we first check if every individual quadratic cost is convex, if yes, then we can directly claim the total cost is convex; otherwise we compute the summed cost and check if the summed cost is convex. I think this should create least surprising results to the user, when they provide some quadratic cost and the total cost is convex. (We do pay the additional cost that we need to compute the total quadratic cost for twice, once when we decide if the problem is convex to invoke the right solver, once when we parse the total cost to each solver.)

jwnimmer-tri, 2022-10-17:
I think if `AggregateQuadraticAndLinearCosts` used a struct for its output argument (instead of four separate individual arguments), then we could probably figure out the right technique for caching that result within a MathematicalProgram.

Something like: store it as a `shared_ptr` member field, reset it anytime the costs change, and use an [atomic swap](https://en.cppreference.com/w/cpp/memory/shared_ptr/atomic) in case it's null and we need to recompute it within a `const` MP getter function.  Getting that right probably means adding a "serial number" to the Cost base class (which is incremented anytime the Cost matrices get updated by the user).

hongkai-dai, 2024-06-25:
Decided not to do this but use AggregateQuadraticAndLinearCosts to sum up the linear and quadratic costs.

closed 2024-06-25 by hongkai-dai