RobotLocomotion/drake #18119: CI failure: stubgen timeout

ggould-tri, 2022-10-17:
In https://drake-jenkins.csail.mit.edu/view/Production/job/linux-focal-gcc-bazel-weekly-snopt-release/4/ the invocation of `stubgen` timed out, resulting in several failures:
```
[3:49:18 AM]  ERROR: /media/ephemeral0/ubuntu/workspace/linux-focal-gcc-bazel-weekly-snopt-release/src/bindings/pydrake/BUILD.bazel:789:22: GenerateMypyStubs bindings/pydrake/pydrake/__init__.pyi failed: (Exit 1): stubgen failed: error executing command bazel-out/k8-opt/bin/bindings/pydrake/stubgen --quiet '--package=pydrake' '--output=bazel-out/k8-opt/bin/bindings/pydrake'
[3:49:18 AM]  
[3:49:18 AM]  Use --sandbox_debug to see verbose messages from the sandbox
[3:49:18 AM]  Traceback (most recent call last):
[3:49:18 AM]    File "/media/ephemeral0/ubuntu/workspace/linux-focal-gcc-bazel-weekly-snopt-release/_bazel_ubuntu/cdb0f36101f07a70b5208b088ff281a2/sandbox/linux-sandbox/4355/execroot/drake/bazel-out/k8-opt/bin/bindings/pydrake/stubgen.runfiles/drake/bindings/pydrake/stubgen.py", line 13, in <module>
[3:49:18 AM]      sys.exit(stubgen.main())
[3:49:18 AM]    File "/media/ephemeral0/ubuntu/workspace/linux-focal-gcc-bazel-weekly-snopt-release/_bazel_ubuntu/cdb0f36101f07a70b5208b088ff281a2/sandbox/linux-sandbox/4355/execroot/drake/bazel-out/k8-opt/bin/bindings/pydrake/stubgen.runfiles/mypy_internal/mypy/stubgen.py", line 1831, in main
[3:49:18 AM]      generate_stubs(options)
[3:49:18 AM]    File "/media/ephemeral0/ubuntu/workspace/linux-focal-gcc-bazel-weekly-snopt-release/_bazel_ubuntu/cdb0f36101f07a70b5208b088ff281a2/sandbox/linux-sandbox/4355/execroot/drake/bazel-out/k8-opt/bin/bindings/pydrake/stubgen.runfiles/mypy_internal/mypy/stubgen.py", line 1652, in generate_stubs
[3:49:18 AM]      py_modules, c_modules = collect_build_targets(options, mypy_opts)
[3:49:18 AM]    File "/media/ephemeral0/ubuntu/workspace/linux-focal-gcc-bazel-weekly-snopt-release/_bazel_ubuntu/cdb0f36101f07a70b5208b088ff281a2/sandbox/linux-sandbox/4355/execroot/drake/bazel-out/k8-opt/bin/bindings/pydrake/stubgen.runfiles/mypy_internal/mypy/stubgen.py", line 1418, in collect_build_targets
[3:49:18 AM]      py_modules, c_modules = find_module_paths_using_imports(
[3:49:18 AM]    File "/media/ephemeral0/ubuntu/workspace/linux-focal-gcc-bazel-weekly-snopt-release/_bazel_ubuntu/cdb0f36101f07a70b5208b088ff281a2/sandbox/linux-sandbox/4355/execroot/drake/bazel-out/k8-opt/bin/bindings/pydrake/stubgen.runfiles/mypy_internal/mypy/stubgen.py", line 1452, in find_module_paths_using_imports
[3:49:18 AM]      result = find_module_path_and_all_py3(inspect, mod, verbose)
[3:49:18 AM]    File "/media/ephemeral0/ubuntu/workspace/linux-focal-gcc-bazel-weekly-snopt-release/_bazel_ubuntu/cdb0f36101f07a70b5208b088ff281a2/sandbox/linux-sandbox/4355/execroot/drake/bazel-out/k8-opt/bin/bindings/pydrake/stubgen.runfiles/mypy_internal/mypy/stubutil.py", line 83, in find_module_path_and_all_py3
[3:49:18 AM]      mod = inspect.get_package_properties(module)
[3:49:18 AM]    File "/media/ephemeral0/ubuntu/workspace/linux-focal-gcc-bazel-weekly-snopt-release/_bazel_ubuntu/cdb0f36101f07a70b5208b088ff281a2/sandbox/linux-sandbox/4355/execroot/drake/bazel-out/k8-opt/bin/bindings/pydrake/stubgen.runfiles/mypy_internal/mypy/moduleinspect.py", line 138, in get_package_properties
[3:49:18 AM]      res = self._get_from_queue()
[3:49:18 AM]    File "/media/ephemeral0/ubuntu/workspace/linux-focal-gcc-bazel-weekly-snopt-release/_bazel_ubuntu/cdb0f36101f07a70b5208b088ff281a2/sandbox/linux-sandbox/4355/execroot/drake/bazel-out/k8-opt/bin/bindings/pydrake/stubgen.runfiles/mypy_internal/mypy/moduleinspect.py", line 164, in _get_from_queue
[3:49:18 AM]      raise RuntimeError("Timeout waiting for subprocess")
[3:49:18 AM]  RuntimeError: Timeout waiting for subprocess
[3:49:40 AM]  [10,498 / 10,526] 568 / 5885 tests, 1 failed; Compiling geometry/proximity/test/mesh_half_space_intersection_test.cc; 32s remote-cache, linux-sandbox ... (27 actions, 26 running)
[3:52:16 AM]  [13,386 / 13,412] 898 / 5885 tests, 1 failed; Compiling geometry/test/geometry_state_test.cc; 26s remote-cache, linux-sandbox ... (28 actions running)
[3:55:22 AM]  [17,661 / 17,686] 1879 / 5885 tests, 2 failed; Compiling geometry/proximity/test/triangle_surface_mesh_test.cc; 24s remote-cache, linux-sandbox ... (28 actions running)
[3:58:44 AM]  [23,153 / 23,182] 3051 / 5885 tests, 2 failed; Compiling multibody/contact_solvers/sap/test/sap_model_test.cc; 15s remote-cache, linux-sandbox ... (28 actions running)
[4:02:35 AM]  [28,235 / 28,261] 4276 / 5885 tests, 3 failed; Testing //multibody/fem:petsc_symmetric_block_sparse_matrix_test; 132s remote-cache ... (28 actions, 27 running)
[4:06:57 AM]  [34,093 / 34,119] 5710 / 5885 tests, 3 failed; Testing //multibody/fem:petsc_symmetric_block_sparse_matrix_test; 402s remote-cache ... (27 actions, 25 running)
[4:07:36 AM]  INFO: Elapsed time: 2102.846s, Critical Path: 464.62s
[4:07:36 AM]  INFO: 34728 processes: 17466 internal, 17228 linux-sandbox, 27 local, 7 worker.
[4:07:36 AM]  FAILED: Build did NOT complete successfully
[4:07:36 AM]  //:py/install_test                                              FAILED TO BUILD
[4:07:36 AM]  //bindings/pydrake:py/pyi_sanity_test                           FAILED TO BUILD
[4:07:36 AM]  //tools/install:py/install_test_helper_test                     FAILED TO BUILD
```
Opening and immediately closing this bug as possibly a one-off (we haven't seen this failure before in tests, although we have had relatively few runs to see it in, and it cleared on a re-run).  Reopen this issue if it recurs.

ggould-tri, 2022-11-02:
Recurred in https://drake-jenkins.csail.mit.edu/view/Production/job/linux-focal-unprovisioned-gcc-cmake-nightly-debug/564/ .  Reopening.

SeanCurtis-TRI, 2022-12-06:
Happened again:
https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-focal-unprovisioned-gcc-cmake-nightly-debug/600/

ggould-tri, 2023-01-04:
https://drake-jenkins.csail.mit.edu/view/Production/job/linux-focal-unprovisioned-gcc-bazel-nightly-debug/628/

That's once in in October, November, December, January; whatever this is, it persists with grim regularity.  Bumping the priority by the smallest increment.

svenevs, 2023-01-24:
Another instance today on focal debug nightly, we need to give this more time.

SeanCurtis-TRI, 2023-04-20:
Happened [again](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-focal-unprovisioned-gcc-cmake-nightly-release/1061/). Even 6 minutes isn't enough.

xuchenhan-tri, 2023-05-13:
Happened again [here](https://drake-jenkins.csail.mit.edu/job/linux-focal-unprovisioned-gcc-bazel-nightly-packaging/79/)

BetsyMcPhail, 2023-05-24:
Two instances last night:

https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-focal-unprovisioned-gcc-cmake-nightly-debug/769/
https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-focal-unprovisioned-gcc-cmake-nightly-everything-release/1096/

BetsyMcPhail, 2023-05-30:
May 30th: https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-focal-unprovisioned-gcc-bazel-nightly-packaging/96/

BetsyMcPhail, 2023-06-21:
6/21: https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-focal-unprovisioned-clang-bazel-nightly-everything-debug/799/

svenevs, 2023-08-02:
2023-08-03:

- [linux-focal-gcc-cmake-nightly-everything-debug/839](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-focal-gcc-cmake-nightly-everything-debug/839/consoleFull)
- [linux-focal-unprovisioned-gcc-bazel-nightly-debug/841](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-focal-unprovisioned-gcc-bazel-nightly-debug/841/consoleFull)

=> https://github.com/RobotLocomotion/drake/pull/19916

xuchenhan-tri, 2023-10-18:
Happened again on 2023-10-18

https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-gcc-cmake-continuous-release/4934/
https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-gcc-cmake-continuous-release/4935/

cohnt, 2023-10-18:
Looks like that's what's happening here as well?

https://drake-jenkins.csail.mit.edu/job/linux-focal-gcc-cmake-experimental-release/9784/

BetsyMcPhail, 2023-10-19:
10/19:

[linux-focal-gcc-cmake-nightly-debug](https://drake-jenkins.csail.mit.edu/view/Production/job/linux-focal-gcc-cmake-nightly-debug/916/console)
[linux-focal-gcc-cmake-nightly-everything-release](https://drake-jenkins.csail.mit.edu/view/Production/job/linux-focal-gcc-cmake-nightly-everything-release/1239/console)
[linux-focal-unprovisioned-gcc-bazel-nightly-debug](https://drake-jenkins.csail.mit.edu/view/Production/job/linux-focal-unprovisioned-gcc-bazel-nightly-debug/921/consoleFull)
[linux-focal-unprovisioned-gcc-cmake-nightly-everything-debug](https://drake-jenkins.csail.mit.edu/view/Production/job/linux-focal-unprovisioned-gcc-cmake-nightly-everything-debug/919/)
[linux-focal-unprovisioned-gcc-wheel-nightly-release](https://drake-jenkins.csail.mit.edu/view/Production/job/linux-focal-unprovisioned-gcc-wheel-nightly-release/232/console)
[drake-external-examples](https://drake-jenkins.csail.mit.edu/job/RobotLocomotion/job/drake-external-examples/job/main/312/console)

jwnimmer-tri, 2023-10-19:
This has become a lot more frequent recently.  I'm digging into it more.

Observations so far:

(1) There's no need to ship `*.pyi` files for pure python modules.  We can drop those and conserve some effort.

(2) Stubgen generates files in `$PWD/.mypy_cache`.  Genrules are not supposed to write to `.`.  We'll need to adjust that somehow, but it doesn't _seem_ like this has anything to do with the timeouts.  Nuking the cache doesn't slow anything down.

(3) Stubgen ends up generating `__pycache__` garbage in runfiles.  Ideally we'd disable that, but it doesn't _seem_ like this has anything to do with the timeouts.  Nuking the cache doesn't slow anything down.

(4) Stubgen ends up forking & retrying more than necessary, due to failed imports.  We should disable package search (instead just specifying the exact module list we want), and we should try to figure out what other imports are failing.

tom-osika, 2026-10-22:
10/22/2026: https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-tahoe-clang-bazel-nightly-everything-release/6/consoleFull
10/21/2026: https://drake-jenkins.csail.mit.edu/job/mac-tahoe-clang-bazel-continuous-release/12/consoleFull

tyler-yankee, 2026-11-04:
for paper-trail purposes: I don't have links offhand, but we've also seen this issue on a few pre-merge jobs recently.

tom-osika, 2026-11-04:
> for paper-trail purposes: I don't have links offhand, but we've also seen this issue on a few pre-merge jobs recently.

[This](https://drake-jenkins.csail.mit.edu/job/mac-tahoe-clang-bazel-experimental-release/view/change-requests/job/PR-25187/5/) was one of them (for the vtk November upgrade [PR](https://github.com/RobotLocomotion/drake/pull/25187))

PR #18527 (2023-01-04): Increase mypy timeout for glacial CI machines
For now, we'll just bump the timeout and see if that is enough for CI be happy.  If not, possibly the process is stuck forever instead of just slow.

See also https://github.com/RobotLocomotion/drake/pull/17552 and https://github.com/python/mypy/pull/13109 for prior failed attempt (increasing from 5s to 30s).

Closes #18119.

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/18527)
PR #18644 (2023-01-24): Increase mypy timeout for glacial CI machines again
Followup to #18527.  Closes #18119.  Linux debug jobs have been timing out again, increasing from "90 seconds" to "180 seconds" to try and give it enough time to poll the results.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/18644)
commit efac8722 (2023-01-24): Increase mypy timeout for glacial CI machines again (#18644)

Followup to #18527.  Closes #18119.  Linux debug jobs have been
timing out again, increasing from "90 seconds" to "180 seconds"
to try and give it enough time to poll the results.
PR #20403 (2023-10-20): [py tools] Vastly improve stubgen wrapper
Closes #18119 -- at least we hope so.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/20403)