RobotLocomotion/drake #18133: Simulating a cloned context containing hydroelastic contact results fails.

joemasterjohn, 2022-10-18:
### What happened?

While working on #18057 we came across the following:

- Forward simulate a plant with active hydroelastic contacts to time t.
- Clone the context of that plant at time t.
- Construct a new simulator using the cloned context.
- Forward simulate the new simulator with the cloned context.
- Failure at the next timestep when adding hydroelastic contact results (relevant log below)

Reproduction branch with containing the reference commit sha can be found here:
https://github.com/joemasterjohn/drake/tree/hydro_cloned_context_bug

@sammy-tri came up with the proposed fix during review:
```
diff --git a/multibody/plant/contact_results.cc b/multibody/plant/contact_results.cc
index fd2d3925ba..869665b7d6 100644
--- a/multibody/plant/contact_results.cc
+++ b/multibody/plant/contact_results.cc
@@ -66,11 +66,8 @@ void ContactResults<T>::set_plant(const MultibodyPlant<T>* plant) {
 template <typename T>
 void ContactResults<T>::Clear() {
   point_pairs_info_.clear();
-  if (hydroelastic_contact_vector_ownership_mode() == kAliasedPointers) {
-    hydroelastic_contact_vector_of_pointers().clear();
-  } else {
-    hydroelastic_contact_vector_of_unique_ptrs().clear();
-  }
+  hydroelastic_contact_info_ =
+      std::vector<const HydroelasticContactInfo<T>*>();
   plant_ = nullptr;
 }
 ```

### Version

6c7163f15334e1cc407dec3c51b7825347566959

### What operating system are you using?

Ubuntu 20.04

### What installation option are you using?

compiled from source code using Bazel

### Relevant log output

```shell
abort: Failure at multibody/plant/contact_results.cc:98abort: Failure at multibody/plant/contact_results.cc:98 in AddContactInfo(): condition 'hydroelastic_contact_vector_ownership_mode() == kAliasedPointers' failed. in AddContactInfo(): condition 'hydroelastic_contact_vector_ownership_mode() == kAliasedPointers' failed.
```

SeanCurtis-TRI, 2022-10-18:
Is this possibly related to #17375.

jwnimmer-tri, 2024-11-02:
Best guess, I fixed this when I panzered all of the hydro contact storage in #21623 and its predecessors.

closed 2024-11-02 by jwnimmer-tri