RobotLocomotion/drake #18150: cmake: Should provide less fragile way to extract PYTHONPATH for installed CMake Drake?

EricCousineau-TRI, 2022-10-20:
**Is your feature request related to a problem? Please describe.**
Relates workflow described in https://github.com/RobotLocomotion/drake-ros/issues/157

**Describe the solution you'd like**
Ideally, we provide something like `{module}_PYTHONPATH`?

**Describe alternatives you've considered**
Copy-pasta from `drake-external-examples`, e.g.
https://github.com/RobotLocomotion/drake-external-examples/blob/eabc1a36bd7fe1537aadbb559e6af76ed3878d9f/drake_cmake_external/drake_external_examples/CMakeLists.txt#L38-L50

**Additional context**
N/A

EricCousineau-TRI, 2022-10-20:
Assigning to @jwnimmer-tri as component owner for delegation etc.

jwnimmer-tri, 2022-10-20:
What do other CMake packages currently do today in terms of providing the path to their python module(s)?

EricCousineau-TRI, 2022-10-20:
From inspecting current Focal tarball, perhaps we provide something like `pydrake-config.cmake`, or `drake-python-config.cmake`?

> What do other CMake packages currently do today in terms of providing the path to their python module(s)?

Not sure - checking...

EricCousineau-TRI, 2022-10-20:
Checked Open3D, didn't see anything about CMake  + C++ + Python.
At least from briefly perusal, I only see C++ ^ Python for consumption:
http://www.open3d.org/docs/release/compilation.html
http://www.open3d.org/docs/release/cpp_project.html
https://github.com/isl-org/open3d-cmake-external-project/blob/b5b8ed94bb79aed3d78a521c59e40874f286f05d/CMakeLists.txt#L35

For ROS 2 packages, I believe it uses `ament` to consolidate things into the same install tree s.t. all projects install to same `PYTHONPATH` that is then accessible via generated `setup.sh` scripts. (\cc @IanTheEngineer @cottsay to correct me if I'm wrong)
I believe ROS 1 was similar.

So not immediately sure there's a standard way.

jwnimmer-tri, 2022-10-20:
For the ROS users who want pydrake, the answer will be "Use the Drake binaries that are shipped inside the ROS2 apt site" which presumably will place the code onto same Python path as all of the other ROS-Python code and work in the standard and expected ways with ament, so this issue is moot.

The question of this issue would be for CMake users who do _not_ use ROS, i.e., `drake_cmake_external`.

I can ask Kitware folks what they know about best practices in that case.

EricCousineau-TRI, 2022-10-20:
> For the ROS users who want pydrake [...]

Yup, makes sense!

> I can ask Kitware folks what they know about best practices in that case.

Yeah, that'd be great!

svenevs, 2022-10-27:
> perhaps we provide something like pydrake-config.cmake, or drake-python-config.cmake?

That seems like overkill, it would have to depend on `find_dependency(drake)`.  The reason to do this is if you need to export the python targets for cmake consumers to utilize, which doesn't really seem to have a use case that I'm aware of (we're installing libraries not modules).

For this scenario, the existing drake cmake config file installed should just `set(DRAKE_PYTHON_DIR "...")` or any other similarly named `DRAKE_*` variable to your naming desires.  There's no real convention in CMake for this scenario, but making the directory available so that a consumer can set their `PYTHONPATH` accordingly is a nice thing to do.

```cmake
# consuming project
find_package(drake CONFIG REQUIRED)
# DRAKE_PYTHON_DIR variable is populated via `set()` statement in `drake-config.cmake`
# and available for use onward
```

jwnimmer-tri, 2022-10-27:
\CC @EricCousineau-TRI @cottsay FYI on the proposed design (see immediately above).

Best guess, the `DRAKE_YTHON_DIR` would be implemented via a change to `drake/tools/install/libdrake/drake-config.cmake`.  The Python version being used at build-time is accessible in `BUILD.bazel` as something like `load("@python//:version.bzl", "PYTHON_SITE_PACKAGES_RELPATH", "PYTHON_VERSION")`.

We'll probably get to this in ~December or so, but if you'd like to use it sooner feel free to open a PR and I can help review.

EricCousineau-TRI, 2022-10-27:
That proposal sounds good! And acknowledged, will mention in https://github.com/RobotLocomotion/drake-ros/issues/157
Thanks!

jwnimmer-tri, 2022-10-27:
On a related note, we could probably offer `DRAKE_PYTHON_VERSIONS="3.8"` as well, in case the user wants to fail-fast when their Python interpreter version would be unable to load `pydrake`.  Note that it would be a **list** of version numbers in case we ever start providing multi-python binaries.

PR #18201 (2022-10-28): Provide Python module information in drake-config.cmake
This will provide a reliable way for downstream CMake projects to get information about the Python modules installed as part of Drake.

Resolves #18150

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/18201)
closed 2022-10-28 by jwnimmer-tri