RobotLocomotion/drake #18151: RotationMatrix to RPY

paarthshah-tri, 2022-10-20:
**Is your feature request related to a problem? Please describe.**
No explicit problem, just a simple feature request.

**Describe the solution you'd like**
The RotationMatrix class should have a  function .toRollPitchYaw() simliar to the .ToQuaternion() and .ToAngleAxis() functions, especially given that it has a constructor that takes in a RollPitchYaw class.

**Describe alternatives you've considered**
Create a RollPitchYaw function from the RotationMatrix since

ggould-tri, 2022-10-24:
Assigning @rpoyner-tri for now per component ownership; Rico, feel free to reassign.

mitiguy, 2022-10-24:
I talked to @paarthshah-tri early about this and will do it now.

mitiguy, 2022-10-24:
PR #18172 helps to address the C++ part of issue #18151.
A separate PR will create Python bindings to finish resolving that issue.

mitiguy, 2022-10-27:
PR https://github.com/RobotLocomotion/drake/pull/18187 resolves issue https://github.com/RobotLocomotion/drake/issues/18151.

closed 2022-10-27 by mitiguy