RobotLocomotion/drake #18161: doxygen website doesn't scroll horizontally enough to see workspace methods

RussTedrake, 2022-10-23:
### What happened?

I was trying to look up the doxygen for solvers::Solve, and couldn't find it!  I assume it's there at https://drake.mit.edu/doxygen_cxx/namespacedrake_1_1solvers.html, but it was actually impossible for me to tell:
[image]

Note that not only are the method names not visible, but the horizontal scrollbar at the bottom doesn't let me scroll more.

I'm using Chrome on Focal.

### What operating system are you using?

Ubuntu 20.04

jwnimmer-tri, 2022-10-23:
The fix here should be to adjust the API content to render correctly.  The changes stalled at https://github.com/RobotLocomotion/drake/issues/17398#issuecomment-1158337471 might end up improving the scrollbars, but having a 3000-pixel-wide return type on a function is the primary problem.

To remove some of the template goop from API docs, one tool we have is the `#ifndef DRAKE_DOXYGEN_CXX` on the SFINAE stuff.

On the the other hand, `AddLogarithmicSos2Constraint` (one of the too-wide offenders) is overly verbose even in the header itself, not just in Doxygen.  The function should just return `auto` in which case the compile-time size of the vector can be computed directly within the function using `constexpr`, with no helper code.  The shim type `LogarithmicSos2NewBinaryVariables` should simply be deleted (via deprecation).

jwnimmer-tri, 2025-11-17:
FYI This is the same kind of thing as #11317.  It probably makes sense to solve them together.