RobotLocomotion/drake #18177: vSphere x86 CI Matrix and Disk Usage Incompatible

svenevs, 2022-10-25:
[Slack discussion here](https://drakedevelopers.slack.com/archives/C270MN28G/p1666630520267259).

For unknown reasons the disk I/O usage is too slow all of a sudden.  We reduced the instance cap to no avail.  Stopgap: remove `clang-bazel-nightly-debug` and `clang-cmake-nightly-debug`, then begin investigations.

svenevs, 2022-10-26:
MacStadium ticket 143976 filed with known historical data, hopefully it is an easy fix such as hardware replacement.

BetsyMcPhail, 2022-11-08:
I removed some snapshots and did some other datastore cleanup. We are back to about the same baseline datastore usage as all of last year. Note that most of that time period was _before_ the space increase in June, so as a percentage we are lower than before.

I increased the instance cap in Jenkins back to 7.  We'll wait and see how things go tonight.

BetsyMcPhail, 2022-11-10:
Following the plan in #17969:

- mac-x86-monterey-clang-bazel-nightly-debug was added to replace mac-x86-big-sur-clang-bazel-nightly-debug
- all mac cmake debug jobs, including mac-x86-~~big-sur~~monterey-clang-cmake-nightly-debug were removed. 

We're currently only running Monterey jobs on VSphere and they are not showing any slowdown. We can continue to monitor when Ventura is added and re-open this issue if needed.

closed 2022-11-10 by BetsyMcPhail