RobotLocomotion/drake #18195: [geometry] Scalar conversion of ProximityProperties

DamrongGuoy, 2022-10-27:
@amcastro-tri mentioned that ProximityProperties do not allow us to scalar convert to AutoDiffXd, and that's why #18188's unit test only covers `double` but skips AutoDiffXd and [symbolic::Expression](https://drake.mit.edu/doxygen_cxx/classdrake_1_1symbolic_1_1_expression.html). 

At the moment I can neither confirm nor deny.  The doxygen https://drake.mit.edu/doxygen_cxx/classdrake_1_1geometry_1_1_proximity_properties.html doesn't mention anything.

This issue will help me keep track of this issue in ProximityProperties that affect downstream code.

DamrongGuoy, 2022-10-27:
I agree that #18188 doesn't need this anymore. I updated the problem description.

DamrongGuoy, 2022-10-27:
To be clear, @amcastro-tri, by scalar conversion, you meant if we construct properties using this function with template parameter ValueType = `double`:
```C++
  template <typename ValueType>
  void AddProperty(const std::string& group_name, const std::string& name,
                   const ValueType& value);
  ProximityProperties properties_double;
  double double_value=0.0;
  properties_double.AddProperty<double>(group, name, double_value);                   
```
Then, later, we might want to change it to AutoDiffXd like this, correct?
```
  ProximityProperties properties_AutoDiffXd = ...convert from properties_double...
  AutoDiffXd autodiff_value = properties_AutoDiffXd.GetProperty<AutoDiffXd>(group, name);
```

jwnimmer-tri, 2024-03-20:
Possibly a duplicate of #5454?  Anyway, not clear what the problem is.

closed 2024-03-20 by jwnimmer-tri