RobotLocomotion/drake #18217: CI failure 'No space left on device' trying to write to /tmp

BetsyMcPhail, 2022-10-31:
Slack discussion: https://drakedevelopers.slack.com/archives/C270MN28G/p1667141831666679

ggould-tri, 2022-11-02:
Recurred https://drake-jenkins.csail.mit.edu/view/Production/job/linux-jammy-unprovisioned-gcc-bazel-nightly-everything-debug/113/

jwnimmer-tri, 2022-11-15:
Regurgitating information from slack so that it's not buried in a stovepipe:

That Jammy build begins life like so (0.8 GiB free):

```
[2022-10-30T08:18:57.069Z]   ------------------------------------------------------------------------------
[2022-10-30T08:18:57.069Z]   *** Disk usage for /tmp:
[2022-10-30T08:18:57.069Z]  Filesystem      Size  Used Avail Use% Mounted on
[2022-10-30T08:18:57.069Z] /dev/root       7.6G  6.8G  805M  90% /
[2022-10-30T08:18:57.069Z]   ------------------------------------------------------------------------------
```

The matching Focal build begins life like so (8.7 GiB free):

```
[2022-10-30T08:39:09.098Z]   ------------------------------------------------------------------------------
[2022-10-30T08:39:09.098Z]   *** Disk usage for /tmp:
[2022-10-30T08:39:09.098Z]  Filesystem      Size  Used Avail Use% Mounted on
[2022-10-30T08:39:09.098Z] /dev/root        16G  6.9G  8.7G  45% /
[2022-10-30T08:39:09.098Z]   ------------------------------------------------------------------------------
```

We need /tmp to have real space available.  Either Jammy AMI should be upgraded to have a 16G / filesystem (sufficient to resolve this symptom), or we should be mounting a ~10G ramdisk on /tmp instead of using network-backed storage (builds would speed up noticeably).

jwnimmer-tri, 2022-11-17:
This is affecting #18340 with the newly-added prereqs.  Bumping up to high priority.

jwnimmer-tri, 2022-11-17:
Per #18340, please also install `libvtk9-dev` on the new Jammy provisioned AMI in anticipation of that PR most likely landing soon.

BetsyMcPhail, 2022-11-17:
Jammy provisioned has been updated:

```
[11:16:48 AM]    ------------------------------------------------------------------------------
[11:16:48 AM]    *** Disk usage for /tmp:
[11:16:48 AM]   Filesystem      Size  Used Avail Use% Mounted on
[11:16:48 AM]  /dev/root        16G  6.3G  9.1G  41% /
[11:16:48 AM]    ------------------------------------------------------------------------------
```

And `libvtk9-dev` has been installed.

Working on Jammy unprovisioned now.

BetsyMcPhail, 2022-11-17:
And unprovisioned is updated:

```
[11:41:31 AM]    ------------------------------------------------------------------------------
[11:41:31 AM]    *** Disk usage for /tmp:
[11:41:31 AM]   Filesystem      Size  Used Avail Use% Mounted on
[11:41:31 AM]  /dev/root        16G  6.8G  8.6G  45% /
[11:41:31 AM]    ------------------------------------------------------------------------------
```

closed 2022-11-17 by BetsyMcPhail

rpoyner-tri, 2025-05-15:
https://drake-jenkins.csail.mit.edu/job/linux-jammy-unprovisioned-gcc-bazel-experimental-debug/25/ (a fairly speculative PR build) hit "no space left on device".