RobotLocomotion/drake #18233: Wheel CMake Build Improvements

svenevs, 2022-11-01:
Redirect from #18189.  None of these issues are show-stoppers.

- [ ] bzip2 `length modifier 'L' results in undefined behavior or no effect with 'u' conversion specifier [-Wformat]`, try updating version.
- [ ] `libtiff` `Manually-specified variables were not used by the project`, set `JPEG_INCLUDE_DIRS` and `LIBLZMA_INCLUDE_DIRS` (`find_package` finds the right thing because of library roots also specified)
- [ ] `mumps`: longstanding warnings about `Type mismatch in argument 's' at (1); passed INTEGER(4) to LOGICAL(4)`, likely no solution (`gfortran`)
- [ ] arm64 builds littered with `warning: could not create compact unwind for`, no known solution

mwoehlke-kitware, 2023-01-20:
FYI, we're already using the latest bzip2.

jwnimmer-tri, 2023-01-20:
> libtiff Manually-specified variables were not  ...

Fixed in #18630, looks good.

> bzip2 length modifier 'L' results in undefined behavior or no effect with 'u' conversion specifier [-Wformat], try updating version.

Not worth exploring.  The warning seems unlikely to be a problem in practice.

> mumps: longstanding warnings about Type mismatch in argument 's' at (1); passed INTEGER(4) to LOGICAL(4), likely no solution (gfortran)

Not worth exploring.  The warning seems unlikely to be a problem in practice.

> arm64 builds littered with warning: could not create compact unwind for, no known solution

IIRC @williamjallen mentioned a bazel output filter flag that could fix this.  I'm pretty sure it happens on non-Wheel builds as well.

closed 2023-01-26 by jwnimmer-tri