RobotLocomotion/drake #18234: Expression::Expand() fails in debug mode on rational function.

frankpermenter, 2022-11-01:
### What happened?

The following snippet of code generates an exception when compiled in debug mode 
```   
         Variable x{"x"};    
         Expression f = x * pow(1  + 1.0 / x , 2); 
         Expression f_expanded = f.Expand();
```
I suspect there is a bug involving powers of rational functions.

To reproduce error, checkout https://github.com/frankpermenter/drake/commit/2a545b5605f91dfda4806b41a157e1fc560ecd2f and run:

``` 
bazel run  //common/symbolic/expression:expression_expansion_bug_test --config=clang --compilation_mode="dbg"

### What operating system are you using?

Ubuntu 20.04

### What installation option are you using?

compiled from source code using Bazel

### Relevant log output

```shell
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from Basic
[ RUN      ] Basic.Test
abort: Failure at common/symbolic/expression/expression_cell.cc:122 in ExpandMultiplication(): condition 'e2.EqualTo(e2.Expand())' failed.
INFO: Invocation ID: 758d5278-9356-4570-806c-9a4aa166d2b4
```

hongkai-dai, 2022-11-02:
I can't understand this error. I added the unit test in my branch https://github.com/hongkai-pression_expand, and tried to do print out before the error being thrown https://github.com/hongkai-dai/drake/blob/9ebf0a79286dd8cc28f7845d79b04e773d248e03/common/symbolic/expression/expression_cell.cc#L121-L128
```cc
  std::cout << "e1: " << e1 << "\ne2: " << e2 << "\n";
  std::cout << "e1.Expand(): " << e1.Expand()
            << "\ne2.Expand(): " << e2.Expand() << "\n";
  std::cout << "e1 == e1.Expand()? " << e1.EqualTo(e1.Expand()) << "\n";
  std::cout << "e2 == e2.Expand()? " << e2.EqualTo(e2.Expand()) << "\n";
  DRAKE_ASSERT(e1.EqualTo(e1.Expand()));
  DRAKE_ASSERT(e2.EqualTo(e2.Expand()));
```
but I got this print out
```
e1: 1
e2: 1
e1.Expand(): 1
e2.Expand(): 1
e1 == e1.Expand()? 1
e2 == e2.Expand()? 1
abort: Failure at common/symbolic/expression/expression_cell.cc:128 in ExpandMultiplication(): condition 'e2.EqualTo(e2.Expand())' failed.
```
Although it just printed out `e2 == e2.Expand()? 1`, it immediately fails in `DRAKE_ASSERT(e2.EqualTo(e2.Expanc()))`. How can this happen?

hongkai-dai, 2022-11-02:
Thanks Jeremy, but I checked `std::cout << "e2 == e2.Expand()? " << e2.EqualTo(e2.Expand()) << "\n";` and it prints 1. So I literally just checked the condition inside `DRAKE_ASSERT` to be true, and then DRAKE_ASSERT fails with the same condition.

I also print
```cc
  std::cout << "e2.get_kind() == e2.Expand().get_kind(): "
            << (e2.get_kind() == e2.Expand().get_kind()) << "\n";
```
and it prints `e2.get_kind() == e2.Expand().get_kind(): 1`

jwnimmer-tri, 2022-11-02:
Sorry, I guess I didn't read it closely.  I've opened #18247 with the test case so we're all of the same page.  I'll see what I can discover.

jwnimmer-tri, 2022-11-02:
I think maybe you we're flushing stdout, and the error was actually on different values?  In the debugger I'm clearly different values when the assertion trips:

```
(gdb) p e1
$5 = (const drake::symbolic::Expression &) @0x7fffffffd668: {boxed_ = {value_ = 2}}
(gdb) p e1_expanded
$6 = {boxed_ = {value_ = 2}}
(gdb) p e2
$7 = (const drake::symbolic::Expression &) @0x7fffffffd5c8: {boxed_ = {value_ = nan(0x1555555739160)}}
(gdb) p e2.cell()
$8 = (const drake::symbolic::ExpressionCell &) @0x555555739160: {
  _vptr.ExpressionCell = 0x55555571e080 <vtable for drake::symbolic::ExpressionVar+16>, use_count_ = {{ _M_i = 45}}, kind_ = drake::symbolic::ExpressionKind::Var, 
  is_polynomial_ = true, is_expanded_ = true}
(gdb) p e2_expanded
$9 = {boxed_ = {value_ = 2}}
```

jwnimmer-tri, 2022-11-02:
Wow, adding "is expanded" checks in more places shows ticking bombs everywhere.  I'll clean that up and then push it to the PR.

jwnimmer-tri, 2022-11-02:
Here's a better error message:

> Expression
> (1 + 2 * (1 / x) + (1 / pow(x, 2)))
> was supposed to be an expansion fixpoint but changed to
> (1 + (1 / pow(x, 2)) + (2 / x))
> after another expansion

hongkai-dai, 2022-11-03:
I see, I think we would prefer `2/x` over `2 * (1/x)` after expansion?

jwnimmer-tri, 2022-11-03:
Yeah, somewhere isn't reaching the fixpoint correctly.

This makes the error go away, but we need to think about the right fix:
```patch
--- a/common/symbolic/expression/expression_cell.cc
+++ b/common/symbolic/expression/expression_cell.cc
@@ -146,7 +146,7 @@ Expression ExpandMultiplication(const Expression& e1, const Expression& e2) {
     for (const pair<const Expression, double>& p : m1) {
       fac.AddExpression(ExpandMultiplication(p.second, p.first, e2));
     }
-    return fac.GetExpression();
+    return fac.GetExpression().Expand();
   }
   if (is_addition(e2)) {
     //   e1 * (c0 + c1 * e_{2,1} + ... + c_n * e_{2, n})
```

hongkai-dai, 2022-11-03:
To make sure we are on the same page, what the current code does when expanding `pow(1 + 1.0/x, 2)`, is that it builds a `ExpressionAddFactory fac`, It first adds (1 + 1.0/x) * 1 to `fac`, which contains the term `(1.0/x)`. It again adds (1 + 1.0/x) * (1.0/x) to `fac`, which again contains the term (1.0/x). Now in `fac` it contains two `1.0/x`. Then when we call `fac.GetExpreesion()`, it just does adds these two terms as `2 * (1.0/x)`, but don't expand this added term. On the other hand, if we call `fac.GetExpression().Expand()` which expands `2 * (1.0/x)`, then it expands this term to `2./x`.

I think changing `fac.GetExpression().Expand()` solves the problem (we also this line as well https://github.com/RobotLocomotion/drake/blob/657c2244c235543cc35aacfb7f06f7bcfa241669/common/symbolic/expression/expression_cell.cc#L134).

Some potential questions:

`fac.GetExpression()` might be a large expression, and we will expand each term in this large expression. A lot of the terms have been expanded, so it is just a matter of checking `is_expanded() == true` and do nothing to these terms. I hope this checking is quick. On the other hand, at this stage, we only need to expand a term if we detect that term is repeated. So is it OK if we change `ExpressionAddFactory::GetExpression()` function? This function could accept an optional argument `GetExpression(bool expand = false)`

jwnimmer-tri, 2022-11-03:
Your explanation of the situation sounds correct.

For a solution, I don't think we can approach it by adding `Expand()` too much, that will (probably?) lead to super-linear slowdowns.  I think we need to carefully consider the invariants and pre/post-conditions of these functions, and figure out where things start to go wrong, and then fix it directly where it happens.

My intuition is that `fac.AddExpression()` will need to be told in this case that (1) its argument is already expanded and (2) we want it to produce an expanded result at the end, so as it goes it needs to adjust how it collects its terms.

hongkai-dai, 2022-11-03:
Reading this part of the code https://github.com/RobotLocomotion/drake/blob/657c2244c235543cc35aacfb7f06f7bcfa241669/common/symbolic/expression/expression_cell.cc#L681-L694, I wonder if we should set `is_expanded_=false` at the end of this `if` branch. Namely we do
```cc
  if (it != expr_to_coeff_map_.end()) {
    // Case1: term is already in the map
    double& this_coeff{it->second};
    this_coeff += coeff;
    if (this_coeff == 0.0) {
      // If the coefficient becomes zero, remove the entry.
      // TODO(soonho-tri): The following operation is not sound since it cancels
      // `term` which might contain 0/0 problems.
      expr_to_coeff_map_.erase(it);
     // Note that we leave is_expanded_ unchanged here. We could imagine
     // inspecting the new expr_to_coeff_map_ to compute a more precise value,
     // but it's not clear that doing so would improve our overall performance.
    } else if (!is_variable(it->second)) {
      is_expanded_ = false;
    }
  } else {

```
if `this_coeff != 0` and the expression is not a variable., then we always say `is_expanded_` is false, as now we have a non-zero coefficient multiplies with a non-variable expression, and this product has not been expanded.

hongkai-dai, 2022-11-03:
> My intuition is that fac.AddExpression() will need to be told in this case that (1) its argument is already expanded and (2) we want it to produce an expanded result at the end, so as it goes it needs to adjust how it collects its terms.

I have a different opinion. I think `fac.AddExpression()` should still hold its current value. In this case it holds the mapping from the expression `1/x` to the coefficient `2`. This expression `1/x` has already been expanded. I think the problem occurs when we call `fac.GetExpression()` https://github.com/RobotLocomotion/drake/blob/657c2244c235543cc35aacfb7f06f7bcfa241669/common/symbolic/expression/expression_cell.cc#L656-L670, which stores the expression as ExpressionAdd, that multiplies the coefficient `2` with the expression `1/x`, leading to the un-expanded expression `2 * (1.0/x)`. So we should modify `GetExpression()` function such that this ExpressionAdd stores the coefficient `1` and the expression `2/x`.

jwnimmer-tri, 2022-11-03:
In this case I believe `ExpressionAddFactory::GetExpression().is_expanded()` will be false (with the current code) because `is_expanded_` is already false (because of the division term).  So the `GetExpression` function is meeting it's contract.

There will be many more examples beyond just `2 / x` where we need to preserve expanded form.  A little band-aid for `2 / x` will not be enough.  If we need an expanded result from the factory, then we either need to call `fac.GetExpression.Expand()` (which might be really slow) or we need to set an option like `fac.always_expand(true)` within `ExpandMultiplication` only, and then change the factory's implementation to always expand and return `is_expanded() == true`, efficiently, when opted-in like that.

Your proposal is tantamount to having the AddFactory factory _always_ unconditionally expand its result, which is both expensive (when people don't need it) and wrong (when people don't want it).

I suppose one quick way out might be to disprove my hypothesis that `fac.GetExpression().Expand()` is too slow.  We could try adding `Expand()` where necessary (more than just one line!) within `ExpandMultiplication` and run the benchmarks, and see what it looks like.

hongkai-dai, 2023-02-14:
Sorry I haven't worked on this for a while. It affects Frank's code so I should solve it (sorry Frank).

We have proposed some band-aid for `2/x`, but that doesn't seem like a good solution. It could slow down expanding other expressions.

I think currently there are ambiguity on how the "expand" operation works. What is the mathematical definition of "Expand"? Here is my attempted answer:
The arithmetic operations `+, - *, /, pow` have precedence rules, that `pow` is takes precedence over `*` and `/`, which also take precedence over `+` and `-`. To change the order of the arithmetic operation, we use parenthesis. For example in `(1+x)ⁿ`, we will evaluate the low-order operation 1+x first, before evaluating the high-order operation `pow`, due to the parenthesis. We expect that after expansion, the expression always evaluates the high-order operation before the low-order operation, namely all the parentheses are removed. To put it simple, we regard "expand" function as "removing parentheses".

I think we should list out some expected behavior of `Expand` function, and then code this function according to the expanded behavior.

Here are some properties that I _think_ we will all agree:
1. The fixed-point property. `e.Expand().Expand() == e.Expand()`.
2. Expand addition (e1 + e2).Expand() = (e1.Expand() + e2.Expand()).Expand().
3. Expand multiplication (e1 * e2).Expand() == (e1.Expand() * e2.Expand()).Expand()
4. Expand power: pow(e, n).Expand() = (pow(e, n-1).Expand() * e.Expand()).Expand(). This is a special case of expanding multiplication.
5. Expand division: (e1 / e2).Expand() = e1.Expand() / e2.Expand().

Here are some of the operations that I _think_ worth discussion:
1. (e1 * (e2 / e3)).Expand(), I think we agree that this should return (e1 * e2).Expand() / e3.Expand(), because `*` and `/` takes the same precedence, so after removing the parenthesis the expression should evaluate from left to right, hence we evaluate e1 * e2 first, before dividing with e3. But this raises the issue if we change the order, namely consider ((e2/e3) * e1).Expand(). Should we return (e2 * e1).Expand() / e1.Expand()? Or we just do ((e2/e3).Expand() * e1.Expand()).Expand()? Namely consider (y/x) * 2, should we return 2y / x or y/x * 2? Both correspond to the same mathematical operation and neither one has parenthesis.

cc @jwnimmer-tri

jwnimmer-tri, 2023-05-23:
I think it would help to be clear _to what use_ we put an expanded expression.

Is the intention to produce a mathematically canonical form?  For example, if we have an algorithm trying to cancel or collect redundant terms, it might help to put all of the terms into a canonical form so that we can be sure to cancel or collect everything that is the "same".

Or, is the intention to produce a "familiar" simplified form?  For example if we are going to display something to the user, there are typical ways to denote a given form that a user would be familiar with, and we should prefer to use those to make reading the output easier.

Or, it it for performance, a kind of "garbage collection"?  By multiplying out and opportunistically collecting similar terms, we can reduce the storage / representation requirement so that code runs faster.  This is somewhat like the "canonical form" use case, except that it doesn't need to be perfect; it's just an optimization for speed.

As currently documented ("Expands out products and positive integer powers in expression.") it is none of those options, but I suspect that we're trying to use it under one of those contracts, even though none of them is promised by the API.