RobotLocomotion/drake #18236: sap, tamsi: Instabilities and unexpected error messages?

EricCousineau-TRI, 2022-11-01:
### What happened?

In Anzu PR 9768, I've documented some odd controller failure cases I encountered when trying to implement some very simple controllers. I have ported these cases to #18235 (permalink: b495191dd9).

From talking w/ @amcastro-tri, these may or may not be unexpected.
However, from my naive perspective, the stability margins are a bit hard for me to grok.

I've tuned the examples in #18235 to only have slight changes in time step, gains, etc.
The controllers are also "dirt" simple, in that they only operate on velocity; no gain scaling via mass matrix (e.g. inverse dynamics).

\cc @xuchenhan-tri @joemasterjohn @sherm1 

### Version

c277b39826ad10a2222dcdb6c32eefafce0bf9f5

### What operating system are you using?

Ubuntu 20.04

### What installation option are you using?

compiled from source code using Bazel

EricCousineau-TRI, 2022-11-23:
There is an odd platform difference in behavior, captured in Anzu PR 9907. (\cc @mitiguy)

closed 2024-11-02 by jwnimmer-tri

amcastro-tri, 2024-11-18:
I'll consider this a dupicate of #18241. I know there are some small differences, but the physics-relevant issues seems to be #18241 only. 
Feel free to reopen if this is not correct @EricCousineau-TRI