RobotLocomotion/drake #18241: sap: Instabilities for hydroelastic-with-fallback for cube-cube contact?

EricCousineau-TRI, 2022-11-02:
### What happened?

This is separate from #18236 in that there is only contact involved here, no control (or gravity).
I am unsure if these two issues are related.

In Anzu PR 9768, I've added a contact case that I encountered when doing haptic control stuff and piping reaction forces back to a GFF haptic device.
I've added these cases to #18235 (permalink 7264e9d849)

This just instantiates two models that are somewhat representative of what I was using in Anzu; one is rigid, the other is compliant.
It then "throws" one box at the other.

Even with a time step of 0.5ms, it can gain an "unexpected" amount of energy.
(If the time step is 1ms, then it "explodes").

I did not test w/ TAMSI.

\cc @xuchenhan-tri @joemasterjohn @sherm1 @DamrongGuoy @siyuanfeng-tri 

### Version

c277b39826ad10a2222dcdb6c32eefafce0bf9f5

### What operating system are you using?

Ubuntu 20.04

### What installation option are you using?

compiled from source code using Bazel

DamrongGuoy, 2022-11-02:
Thank you @EricCousineau-TRI for setting up the example.  I confirm that I saw `bazel run //examples/instability:py/contact_instability_test` printed this:
```
Running tests...
----------------------------------------------------------------------
(0.0, None)
(0.0005, 'sap')
(0.0004, 'sap')
.
----------------------------------------------------------------------
Ran 1 test in 3.320s
```
with this movie of the three tests with time steps of 0, 0.5ms, and 0.4ms.

https://user-images.githubusercontent.com/42557859/199376661-e7d9ee0a-f230-474f-82c3-3c9395d13e8e.mp4

It looks to me like the research problem of "Artifacts at large time steps" (or high speed impact) that @joemasterjohn is working on.

xuchenhan-tri, 2023-01-12:
This fell through the crack for me before. I wouldn't expect this to go unstable even with the large hydro modulus so I dug to see what caused it.

Turns out the estimated phi0 is bad. Here (rigid hydro vs. compliant hydro), we estimate phi0 as p/g where p is the pressure and g is the pressure gradient in the direction of the contact normal. When the contact normal is close to being orthogonal to the pressure gradient, we get a gigantic estimate of phi0. In this case, it turns out to be on the order of -1e12 when I use dt = 1ms. This creates a crappy bias velocity for SAP (and shouldn't affect TAMSI IIRC).

EricCousineau-TRI, 2023-01-18:
Thank y'all!

Xuchen - I think I understand some of that :sweat_smile:
Is the action item here "Nothing yet, but consider better formulation"?
Additionally, is there anything like "Add a warning/error if this bad estimate crops up"? (is there a chance this warning / error could be false-positive?)

xuchenhan-tri, 2023-01-18:
We discussed this in the softsim meeting. I'd love to see a better formulation and/or a better hydro pressure field for the box, but after fiddling with it for a while, it doesn't seem easy.

Two thoughts on alleviating this issue in the short term:
1. Reducing the hydro modulus for your compliant geometry. 1.8e11 is unnecessarily high.
2. We have a numeric check that prevents contact point being added when the pressure gradient is orthogonal to the contact normal. It might help if we are more aggressive in culling out contact point whose pressure gradient is close to being orthogonal to the contact normal.

jbinagia, 2023-02-07:
Hi all, I just wanted to write that I spoke with @DamrongGuoy about an issue we are seeing at Amazon for box-on-box hydroelastic contact between a rigid and compliant object and that he mentioned it is possibly related to this open issue. I can provide more detail if it is helpful in debugging the issue, but essentially what we had was:
* We were dropping a compliant hydroelastic object with modulus 1.6e7 from a short height onto a rigid hydroelastic object (a conveyor belt, whose collision geometry is a box) 
* We are using the SAP solver

We find that everything works fine at 1 ms. In exploring large time steps (5 ms), we found that quite a large contact force is produced and the package deflects off the conveyor quite violently. Everything works fine if we make both objects compliant (for both 1 ms and 5 ms). Let me know if there is any other info that would be helpful to provide.

amcastro-tri, 2023-02-08:
Hello @jbinagia. The problem you describe seems pretty much the problem reported in this issue. 
The underlying problem is that at large time steps objects can get into a state of collision even before the contact solver can do anything about it. Once in penetration, the solver must now resolver large impulses to takes these objects out of this state. We are currently working on better solutions. For now, the only thing a user can do is to either reduced the time step (at the expense of a more expensive simulation) or even make their objects more compliant (a bit of a hack, but should at least alleviate the problem for now).

jbinagia, 2023-02-08:
Hi @amcastro-tri that makes perfect sense to me (in regards to the need to resolve large impulses at this high time step value). Right now we're mostly using a 1 ms time step (your first suggestion of reducing the time step) so things are working fine, but if we start to move towards higher time steps for some of our configurations I'll switch objects over to `compliant` with a modest modulus. Let me know though if you do end up merging a new solution since in that case it would probably be easiest just to use `rigid` across the board for those object. Thanks again for the help!

amcastro-tri, 2023-02-08:
We'll definitely keep you updated. This is something at the very top of our priority list.

DamrongGuoy, 2023-02-10:
I'm following the ideas from @xuchenhan-tri to check whether this is a problem:
1. Let grad_pressure be the compliant object's pressure gradient. Usually, grad_pressure is near orthogonal to the contact patch. Intuitively the pressure gradient tries to *push* the contact patch like this `|⤑`.
2. In @EricCousineau-TRI 's example, there are some grad_pressure that are near tangential to the contact patch (around the rim of the contact patch). 
3. Intuitively that pressure gradient tries to *slide* along the contact patch like this  `|↑`.
4. Can this give us trouble? As an early investigation, I set up #18765 to show that we can get these quantities in a unit test of a contrived example:
[image]

It remains to be seen how these numbers are troublesome:
- typical 5e2 Newtons (undamped) normal force
- unusually low effective stiffness 5e-10 Pa*m
- super long **1 terameters** (1e12 meters) approximated signed distance
- They are `f_n,e`, `k`, and `φ_0` in our [discrete hydroelastics paper](https://arxiv.org/pdf/2110.04157.pdf).

DamrongGuoy, 2023-02-10:
In addition to the "wrong" pressure gradients, I do believe the classical pass-through problem (induced by large time step or fast motion) is relevant too. I encourage @joemasterjohn to continue in that direction too.

amcastro-tri, 2023-02-15:
We also talked about some low hanging fruit tests a couple of weeks ago. The ones I remember:
 1. Run with TAMSI. Purpose: To determine if this is specific to SAP only or a discrete hydro inherent problem.
 2. For 1, making damping zero in both TAMSI and SAP, should make the models equivalent, for a better apples to apples.
 3. Modify the [hard-coded angle check in hydro](https://github.com/RobotLocomotion/drake/blob/f062738a338ab26776d24c14e6d5d79b2280a79c/geometry/proximity/field_intersection.cc#L145) to avoid pairs with angles larger than 90 degrees. To either verify  or rule out the hypothesis of the angle between gradient and normal being a problem.

Let me know if you need help with any of those. They should allow us to at least answer two questions:
 1. Is this discrete contact (I am pretty sure this will happen with point contact also) or SAP specific?
 2. Is the angle between pressure gradient and surface normal problematic?

DamrongGuoy, 2023-02-16:
Thank you. I have done most of those.

TAMSI doesn't seem to have the problem. However, I used the damping set in the given input deck. I'll have to run again with zero damping.

Modifying the [hard-coded angle check in hydro](https://github.com/RobotLocomotion/drake/blob/f062738a338ab26776d24c14e6d5d79b2280a79c/geometry/proximity/field_intersection.cc#L145) did avoid the problem.  This supports the hypothesis of the angle between gradient and normal being a problem.

We also talked about setting the initial condition that the small compliant box is already penetrating the big rigid box at a small distance, say 1 millimeter.

To help us test all the variations conveniently, I have set up a Jupyter notebook to show what I found. I'll present it in the next team meeting.

https://github.com/DamrongGuoy/drake/blob/BoxBoxNotebook/tutorials/box_box.ipynb

At the moment, I wrote this summary at the end of that Jupyter notebook:

[image]

jwnimmer-tri, 2024-11-02:
@amcastro-tri Possibly this issue is worthy of an updated state-of-affairs posting?  I imagine that the opportunity to add `margin` impacts the discussion here.

xuchenhan-tri, 2024-11-06:
I don't think margin would help this problem in particular. @AdityaBhat-TRI's anzu sim (unfortunately can't link here) seems to be another instance of this issue. We should be able to observe the issue with Damrong's notebook above, with the margins added.