RobotLocomotion/drake #18260: value: Semantics for Value<T> along inheritance hierarchies can be awkward

EricCousineau-TRI, 2022-11-03:
**Is your feature request related to a problem? Please describe.**
In Anzu PR 9782, and related Drake PR #18257, it was a bit odd to figure out (a) what the right form of `Value<T>` was to use along a type hierarchy, and (b) our usage of `dynamic_cast` coupled with `OutputPort.Eval<LowerType>(context)` can be a bit odd.

**Describe the solution you'd like**
Perhaps just better error message?
A bit constrained on time atm, but can provide example as pointed out in overview of Anzu PR.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A. All context provided above.

EricCousineau-TRI, 2022-11-03:
@jwnimmer-tri Not sure who the owner is here. Can I ask who may be the point person to delegate?

jwnimmer-tri, 2022-11-04:
I'm a bit torn about what to do here.

The fundamental problem that you ran into is that we've exposed `ContextBase` (and `SystemBase`) to users.  Having those as non-template classes for implementation clarity and build-system performance was fine, but they should have been placed within an internal namespace (with documentation on public templated class, like we do today with the MbT <=> MbP twins).  Hindsight is 20-20.

However, the particular difficulties mentioned in this ticket aren't really the right angle to elucidate and attack _that_ problem, though.  (The website API references' disfigurement is a more critical aspect.)

One answer to "Doctor, it hurts when I use `Value<T>` on disfigured types" is "Don't do that".  So maybe we should improve the documentation somewhere?  Or better, try to automatically detect the problem and fail-fast?  Or try to teach Value how to type-erase classes with a shadowed Clone, e.g., the user needs to add a member typedef with the base type?

All of that might be possible.  I just find it kind of disappointing to consider investing in any of that, because it's just band-aids that are solving the wrong problem.

Maybe we should just suck it up and mark the base types as internal, even though that's a pretty big hammer?  It would pound a lot of nails, though.

EricCousineau-TRI, 2022-11-04:
> The fundamental problem that you ran into is that we've exposed [...]

I don't really agree. That is part of the problem. But the full type hierarchy that I'm struggling with is
`ContextBase -> Context<T> -> LeafContext<T>`

Sealing `ContextBase` only hurts my user story. I disagree w/ general thrust of concealing `ContextBase`, `SystemBase`. They are publicly inherited portions of public interfaces.

> However, the particular difficulties mentioned in this ticket aren't really the right angle to elucidate and attack that problem, though. [...]

I sort of agree.
The workflow #18257 is related to being able to pass around general information in a `Context`.

If the desire is to tell users, "don't stuff type erased junk inside `Value<>` because you're going to suffer double indirection of type erasure", then I may side with that.

If that is our policy, then we should *strictly* prevent `Value<>` from accepting anything that is virtual.
This is a bit awkwardly at odds between admitting `Clone()`, which is a mechanism for both non-moveable / non-copyable types, part of which are virtual / dynamically polymorphic types (hence the confusion in user workflow).

I am happy to prototype that policy.
Then the solution for users is "don't do `Value<ContextBase>` directly; make wrapper type that *contains* your stuff, then bind that as a `Value<MyWrapperType>`".

jwnimmer-tri, 2022-11-04:
> They are publicly inherited portions of public interfaces.

The user-facing functions on SystemBase like `num_input_ports()` should appear to users as-if they are on `System`.  The fact that some functions are directly inherited across scalar types is an implementation detail.  Having the API split across two (or more) documentation pages is a usability hurdle.  What benefit does it gain for users to expose that split?

> If the desire is to tell users, "don't stuff type erased junk inside `Value<>` because you're going to suffer double indirection of type erasure", then I may side with that. If that is our policy, then we should strictly prevent `Value<>` from accepting anything that is virtual.

I think `Value<geometry::Shape>` probably works fine, and I wouldn't want to disable that.  Is uses inheritance, the static return type of Clone is always the base class.

The only problem hitting you is the conflicting assumptions about what `Value` thinks a `Clone` function should look like (the return type is the base class), and what `Context<T>` implements for `Clone` (return type is not the base class).

ggould-tri, 2022-11-07:
Added component and default assignment.  @sherm1 feel free to reassign, e.g. to the commentators above :-)

EricCousineau-TRI, 2022-11-13:
> The user-facing functions [...] What benefit does it gain for users to expose that split?

I guess the main convenience is less object code / boilerplate otherwise needed to accommodate an artificial split in interfaces. So sure, not that strong, but I don't think the needle to tips the other way.

Consistency does sound better (e.g. make it `Context<T> Clone()`), but not sure if aggressive encapsulation is requisite.

> I think Value<geometry::Shape> probably works fine, [...]

Interesting. This means C++ and Python users may need to be aware of the two levels of type erasure.
That's fin by me. Ultimately, as both a user and a developer, I want there to be less chance for error when using `Value<T>` and/or `AbstractValue::Make<T>()` in both C++ and Python, and (strict) guidance towards the correct pattern.

In this case, I'd be happy to leverage `ShapeBase` as a concrete case study. 
Concretely, for the above, I would like to posit the following:
- `Value<ShapeBase>` is the only correct spelling, ever. `Value<DerivedShape>` is wrong, and we should fail fast if anyone tries ot use it (C++ or Python). Users should explicitly use `Value<ShapeBase>`, and should avoid certain cases of `AbstractValue::Make<>()` due to sharp edges due to type inference.
- We should always and forever prevent direct aggregates in the case of `Value<ShapeBase>`, e.g. `vector<unique_ptr<ShapeBase>.` is not permissible, ever. Use a wrapper class.
I believe all of this is easily achievable in both C++ and Python.

I believe the lessons learned here can easily apply towards `Context<T>`, both in terms of disabling / redirecting away from `Value<ContextBase>` and things like `Value<LeafContext<T>>`.

If you agree on these points, or have concrete alternatives, I am happy to prototype a possible solution route.

jwnimmer-tri, 2022-11-13:
I'm not sure what `ShapeBase` is supposed to be.  Is that just a typo for `Shape`?

EricCousineau-TRI, 2022-11-14:
Ah, yup.

sherm1, 2022-11-14:
Assigning to @jwnimmer-tri for disposition.

closed 2026-09-10 by jwnimmer-tri