RobotLocomotion/drake #18286: macOS arm64 CI cache server

svenevs, 2022-11-09:
Prototype on orka CI for arm64 the cache server.  Relates: #17210.

- MacStadium is developing macOS and linux cache server prototypes that we will be able to test soon.  [Ticket 142250](https://portal.macstadium.com/support-center/tickets/SERVICE-142250).
- TODO: utilize / test / benchmark.  @jwnimmer-tri you mentioned in one of the meetings some specific ideas on testing if I recall, does that ring any bells?

jwnimmer-tri, 2022-11-09:
I don't recall any specific testing ideas.  Best is probably just to point a bazel build at the cache server and see what happens.  Run it once to populate the cache, then a new build (on a different machine) should be very fast (order of ~10x faster).  Those builds could be run by hand, or using Jenkins triggers, either way.

svenevs, 2022-11-15:
Anticipated start date to be able to begin diagnostics and storage analysis: week of 11/30/2022 (after the orka update in #18287).  The hope is to be able to test prototypes of a couple of different flavors of the [cache server options](https://bazel.build/remote/caching#cache-backend).  They appear to almost have a macOS option available but are also looking into getting a linux host connected to try out as well.

svenevs, 2022-12-06:
I found a nice article talking about the bazel cache and it mentions something we may want to look into about the `disk cache`, only skimmed the article for now, putting a bookmark here for later: https://brentley.dev/bazels-remote-caching-and-remote-execution-explained/

svenevs, 2023-01-09:
Whoops!  Sorry for accidentally closing...

Intended to add a followup, in `nginx` land we upload to `${server}:${port}/${build_salt}` where the `build_salt` is just a [hash of various components of the build](https://github.com/RobotLocomotion/drake-ci/blob/main/driver/configurations/cache.cmake).

In [`bazel-remote` land](https://github.com/buchgr/bazel-remote#http11-rest-api) discussing:

> Cache entries are set and retrieved by key, and there are two types of keys that can be used:
>
> 1. Content addressed storage (CAS), where the key is the lowercase SHA256 hash of the entry. The REST API for these entries is: `/cas/<key>` or with an optional but ignored instance name: `/<instance>/cas/<key>`.
> 2. Action cache, where the key is an arbitrary 64 character lowercase hexadecimal string. Bazel uses the SHA256 hash of an action as the key, to store the metadata created by the action. The REST API for these entries is: `/ac/<key>` or with an optional instance name: `/<instance>/ac/<key>`.
>
> ...
>
> **If the `--enable_ac_key_instance_mangling` flag is specified and the instance name is not empty, then action cache keys are hashed along with the instance name to produce the action cache lookup key. Since the URL path is processed with Go's [path.Clean](https://golang.org/pkg/path/#Clean) function before extracting the instance name, clients should avoid using repeated slashes, ./ and ../ in the URL.**

@jwnimmer-tri when you read that do you agree that this instance mangling flag is what we're looking for?  My previous attempt overlooked needing to launch the server with this flag and I'll be able to test that once the current builds finish.  Otherwise, you had mentioned in the meeting that there may be some cache specific flags for salting or doing something like `--action_env=cache_salt=${salt}`?  Did you ever find those flags?

jwnimmer-tri, 2023-01-11:
I think we need even the CAS cache to be partitioned by the drake-ci `DASHBOARD_REMOTE_CACHE_KEY`.  When two builds differ by the GCC version, we need them to use separate caches, because with the way we have Bazel configured, it won't notice the different minor or patch versions.

If there isn't any better flag for it, we can pass `--action_env=REMOTE_CACHE_KEY=...` to mix our key into every action.

jwnimmer-tri, 2023-01-12:
From f2f: let's just add `action_env` noted above for all CI-cached builds.  That should be safe in all cases.

(It actually wouldn't hurt to add everywhere, other than possibly confusing us maintainers reading the logs.)

svenevs, 2023-01-19:
First, some metrics.  Analysis to follow in a summary paragraph -- quick disclaimers about the benchmarks:

- Some of these jobs were sampled a couple of times and averaged.  Some jobs data was only collected once.
- Averaged builds included two versions:
    - A node with both VMs deployed, but one was idle.
    - A node with both VMs deployed, and both running at max capacity.
    - Example: debug everything cache read build took ~2345 seconds to read the cache when there was an idle VM, or ~3445 seconds when both VMs were running full blast.

Takeaway: the metrics are good estimators, but the timings and "speedup" metrics have some pretty large "margins of error".  Some measurements taken early on may be slower than true for CI, I learned at one point that attaching a "ScreenShare" session would crash debug builds (likely too little RAM available) but worked via SSH.  Some release benchmarks were taken via "ScreenShare" though.

Job categories:

- Populate: how long a clean build / test takes to fully populate an empty cache server.
- Isolated: how long it takes to do a clean build / test without any cache server uploading.
- Cached: how long it takes to do a clean build / test reading from the already populated cache.

## `clang-bazel-experimental-release` on `nginx` server

```console
bazel test --test_timeout=300,1500,4500,-1 --keep_going --config=clang --compilation_mode=opt //...
```

| Job      | Time (minutes) | Time (seconds) |
|:--------:|:--------------:|:--------------:|
| Populate | 52.1198        | 3127.19        |
| Isolated | 50.6997        | 3041.98        |
| Cached   | 4.9054         | 294.3267       |

* Cache Size: **11.85935831 GB**
* Speedup (cached / isolated): **9.67549644%**

## `clang-bazel-experimental-release` on `bazel-remote` server

```console
bazel test --test_timeout=300,1500,4500,-1 --keep_going --config=clang --compilation_mode=opt //...
```

| Job      | Time (minutes) | Time (seconds) |
|:--------:|:--------------:|:--------------:|
| Populate | 47.4333        | 2846.0         |
| Isolated | 50.6997        | 3041.98        |
| Cached   | 3.2917         | 197.5          |

* Cache Size: **3.45825966 GB**
* Speedup (cached / isolated): **6.49248187%**

## `clang-bazel-experimental-everything-release` on `bazel-remote` server

```console
bazel test --test_timeout=300,1500,4500,-1 --keep_going --config=clang --compilation_mode=opt --config=everything //...
```

| Job      | Time (minutes) | Time (seconds) |
|:--------:|:--------------:|:--------------:|
| Populate | 61.7856        | 3707.135       |
| Isolated | 59.3333        | 3560.0         |
| Cached   | 18.7762        | 1126.5725      |

* Cache Size: **3.46907305 GB**
* Speedup (cached / isolated): **31.64529494%**

## `clang-bazel-experimental-debug` on `bazel-remote` server

```console
bazel test --test_timeout=300,1500,4500,-1 --keep_going --config=clang --compilation_mode=dbg //...
```

| Job      | Time (minutes) | Time (seconds) |
|:--------:|:--------------:|:--------------:|
| Populate | 139.2093       | 8352.56        |
[... 112 lines cut ...]

After resetting, just using root url http://10.221.188.13:6060

- macOS `du` size after first run: 85166 ~ **41.5849609375 MB**
    - Used no `--action_env=cache_salt=...`
- macOS `du` size after second run: 87483 ~ **42.71630859375 MB**
    - Used `--action_env='cache_salt=salttown'`
- macOS `du` size after third run: 89786 ~ **43.8408203125 MB**
    - Used `--action_env='cache_salt=town-of-salt'`

### File Counts

File analysis (`find /Volumes/orka/cache -type f | wc -l`):

- First run: **1050 files**
    - Used root url: http://10.221.188.13:6060
- Second run: **1072 files**
    - Used root url: http://10.221.188.13:6060
    - Added 22 files.  Seems expected given other results above.
- Third run: **2122 files**
    - Used root url: http://10.221.188.13:6060/supersalty
    - Added 1050 files.  Expected behavior!
- Fourth run: **3172 files**
    - Used root url: http://10.221.188.13:6060/ytlasrepus
    - Added 1050 files.  Expected behavior!

jwnimmer-tri, 2023-01-19:
I've skimmed all of the numbers, but not in a ton of detail.

To me it seems like the summary is that `nginx` and `bazel-remote` are similar in terms of timing (number of seconds for various builds to do their thing).  Maybe 10-30% here or there, but nothing like a 2x factor.

Since nginx is acceptably fast, and seems lower risk (it's what we were doing before, and is a conceptually simpler architecture), it seems like a no-brainer to settle on nginx as the choice?  Is that basically what you're asking?

svenevs, 2023-01-19:
> To me it seems like the summary is that nginx and bazel-remote are similar in terms of timing (number of seconds for various builds to do their thing). Maybe 10-30% here or there, but nothing like a 2x factor.

Correct, they both seem more or less equivalent.  However, where `bazel-remote` really sticks out is the significant reduction in disk space used.  If it is working correctly and as we need it to, if we use `bazel-remote` we could start caching the debug builds for mac as well.

The concern being that if the ActionCache being salted is not enough (the content should also be getting salted), that's not possible.  I took a quick look, in theory it could be added, but I kind of feel like the authors there chose not to for a reason?  In `nginx` land storage doubles, in `bazel-remote` it basically doesn't change with different salts, it is unclear if that is OK or if we'd run into trouble.

> is a conceptually simpler architecture

If anything it's the opposite (with `nginx` we have to implement storage cleanup / cache eviction), but not significantly so.  `bazel-remote` stuck out because it cleans itself and has status checks (could be automated for buildcops), but the main one is cached debug builds.  That said, we currently don't have those so we aren't losing anything by staying with `nginx`, just potentially missing an opportunity to (somewhat dramatically) speedup debug CI.

svenevs, 2023-01-19:
Sorry, final clarification: we're really concerned over the salting as a "why `bazel-remote`" may not be good enough.  If we're over-caching things, bugs introduced now may not even get flagged in CI until weeks or months (whenever the cache gets evicted).  That sounds horrifying to me...

jwnimmer-tri, 2023-01-19:
Could you explain why caching debug builds with nginx (and its larger storage size) would be a problem?  Is is that we have too many Debug variations, and there is some upper-bound on disk space on the cache server storage?  Maybe we should drop support for Everything-Debug on macOS.

Anyway, I think the choice is clear -- nginx is safe and we should do it.

svenevs, 2023-02-07:
Infrastructure docs (closing this issue) are here: https://github.com/RobotLocomotion/drake-ci/pull/198.

BetsyMcPhail, 2023-02-07:
The buildcop documentation should probably be updated to mention the cache warning emails? Also, the do buildcops need to check an ARM job to make sure the cache is up-and-running?

jwnimmer-tri, 2023-03-23:
From f2f -- action for @svenevs here is to write up a list of remaining work (docs, landing the CI PR, etc.).  It's fine if you prefer to open a new issue with those items instead.

svenevs, 2023-04-06:
Finishing the work in #19146, macos side is more or less complete minus some small cleanup that may happen in the unification.

closed 2023-04-06 by svenevs