RobotLocomotion/drake #18300: Packaging staging jobs should not upload .tar.gz archives

svenevs, 2022-11-10:
When testing `//tools/release_engineering/download_release_candidate.py` updates we have been spinning up the associated staging jobs to test what a release would look like.

The `snopt-mosek-packaging` job must create the `.tar.gz` in order to produce the `.deb`, but it does not seem like we should be uploading that `.tar.gz` archive to staging at all (?)

jwnimmer-tri, 2023-01-27:
See #18300 and #14509 for related.

jwnimmer-tri, 2023-02-16:
The action here is probably to change download_release_candidate to use the staging build with the right version number.

jwnimmer-tri, 2023-02-16:
Yes.

What we want here is for the first "download" command to remain unchanged.  Here's the docstring example:
```
  bazel run //tools/release_engineering:download_release_candidate -- \
      --timestamp 20220303 --find-git-sha
```

But then for the second download command (from staging) we _remove_ the timestamp, and download the tgz files from staging instead:
```
  bazel run //tools/release_engineering:download_release_candidate -- \
      --version v1.0.0
```

jwnimmer-tri, 2023-10-05:
BTW Another way of stating the objective:

We want the pages like https://github.com/RobotLocomotion/drake/releases/tag/v1.21.0 to not have YYYYMMDD dates as versions anymore, only ever `1.M.m` in the filename.  (And we should do that by downloading & uploading the Staging artifacts during the release playbook, not by renaming the nightly binaries!)

p.s. Remember to fix the relnotes _template_ as well as part of this; it hard-codes the YYYYMMDD pattern:

https://github.com/RobotLocomotion/drake/blob/3981211b7c534f5e356501412b65ded80bc95943/doc/_release-notes/template.txt#L122

svenevs, 2023-11-07:
FYI staging job does this currently ( https://drake-jenkins.csail.mit.edu/view/Staging/job/linux-focal-unprovisioned-gcc-bazel-staging-packaging/11/consoleFull )

```
[12:27:57 PM]    ------------------------------------------------------------------------------
[12:27:57 PM]    *** CTest Status: UPLOADING PACKAGE ARCHIVE
[12:27:57 PM]    ------------------------------------------------------------------------------
[12:27:57 PM]  
[12:27:57 PM]  -- Uploading drake-20231016-focal.tar.gz to AWS S3...
[12:27:57 PM]  ['/usr/bin/aws', 's3', 'cp', '--only-show-errors', '--acl', 'public-read', '--storage-class', 'STANDARD', '/media/ephemeral0/ubuntu/workspace/linux-focal-unprovisioned-gcc-bazel-staging-packaging/drake-20231016-focal.tar.gz', 's3://drake-packages/drake/staging/drake-20231016-focal.tar.gz']
[12:27:57 PM]  -- Upload complete: https://drake-packages.csail.mit.edu/drake/staging/drake-20231016-focal.tar.gz
[12:27:57 PM]  -- Uploading drake-20231016-focal.tar.gz.sha512 to AWS S3...
[12:27:57 PM]  ['/usr/bin/aws', 's3', 'cp', '--only-show-errors', '--acl', 'public-read', '--storage-class', 'STANDARD', '/media/ephemeral0/ubuntu/workspace/linux-focal-unprovisioned-gcc-bazel-staging-packaging/drake-20231016-focal.tar.gz.sha512', 's3://drake-packages/drake/staging/drake-20231016-focal.tar.gz.sha512']
[12:27:57 PM]  -- Upload complete: https://drake-packages.csail.mit.edu/drake/staging/drake-20231016-focal.tar.gz.sha512
```

PR #20532 (2023-11-14): Update release scripts for staging changes
Staging jobs now produce packages with version numbers rather than timestamps in the file names. Update release instructions and related scripts accordingly.

Fixes #18300.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/20532)
closed 2023-11-14 by mwoehlke-kitware