RobotLocomotion/drake #18303: Document older OS "last supported release"

svenevs, 2022-11-10:
Continuation of: #18294

Discussion points: https://github.com/RobotLocomotion/drake/pull/18299#pullrequestreview-1176539430

Currently notices take place in the release notes, having a single location where the oldest versions supported for e.g., bionic or Big Sur would be helpful to users.

jwnimmer-tri, 2023-01-18:
As a resolution to this, I think we could rename and update this page to something like "Using older operating systems":
https://drake.mit.edu/release_notes/2016.html

It's already linked from the bottom of here:
https://drake.mit.edu/release_notes/release_notes.html

We could probably cross-link it from the installation page too, in a footnote.

mwoehlke-kitware, 2023-01-27:
Should https://drake.mit.edu/release_notes/2015.html also be merged into this new document?

Also, what should it be called? I'm leaning toward `end_of_support.md`?

jwnimmer-tri, 2023-01-27:
> Should https://drake.mit.edu/release_notes/2015.html also be merged into this new document?

Let's say probably not.  It's a different enough document that separate is probably better.

> I'm leaning toward end_of_support.md?

That's fine.  The only other thing that occurs to me is in case we want to allow for its scope growing to more than just literally end of support information, we could call it `timeline.md` or `history.md` something more generic like that.

mwoehlke-kitware, 2023-01-27:
The preamble of both documents is identical. They're both "Using older (something) releases". At the least, it seems they should both be renamed for consistency, e.g. `end_of_support_os.md` and `end_of_support_matlab.md` or similar.

If "scope creep" occurs, we can always rename again.

jwnimmer-tri, 2023-01-27:
Renaming breaks bookmarks, so we should rename as little as possible.  I think we'll need to rename 2016 since it's essentially being rewritten, but I don't see any good reason to touch `2015.md` at all.

PR #18713 (2023-01-31): Consolidate out-of-support documentation
Move end-of-support information to a new, consolidated document. Add a separate section to the release notes content page for "supplemental" pages (i.e. things that make sense here but aren't releases, such as the aforementioned end-of-support information). Replace old documents with a redirect and hide from the contents.

Fixes #18303. ~I'm guessing we will still need to add the EOS information for Bionic.~ Right, information for Bionic *is* added here.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/18713)
closed 2023-01-31 by rpoyner-tri