RobotLocomotion/drake #18311: RenderRpc: Update VTK backend to return a lossless RGB image

zachfang, 2022-11-12:
To achieve consistent label images across different renderers, a render server should return a "magic" RGB image when a label image is requested. The client will then take the RGB image and map it to a label image via the decoding logic in `RenderEngine`.

By doing that, we can also update `server_vtk_backend.cc` to not violate the encapsulation by calling (importing) the protected methods from RenderEngine.

zachfang, 2022-12-26:
We had several discussions regarding this topic in #18344, and I will try to recapture them here to consolidate some of the discussion points.

> (2) the protocol documentation specifies a particular rgb value (either "0,0,0,255 black" or "255,255,255 white") as the background color for label images, and the client learns how to decode / map that special value; in particular it should be a color specification that can stay intact as we change the implementation of label encoding (e.g., if we change labels to use 24 bits instead of just 16 bits), so it should not be documented to its current value (ff, fe, 00, ff).`

&

> If those functions were public, then when we change the Drake implementation from 16-bit labels to 24-bit labels, everyone's existing render servers would break, because the encoding is wrong. (We can't presume that the client and server version pins of Drake are always in lockstep at all times.) In any case, there is no expectation that any given render server will depend on drake or pydrake. We should not require that dependency. In most cases, servers will not depend on Drake, instead depending only on the protocol document. In short, label index <=> rgb bijection must remain as an implementation detail of Drake.

&

>  (1) Every render engine should produce intermediate colored label images that match each other. (So, RPC servers that use a different background color than internal render engines are "bad".)
(2) Hard-coded values are likewise bad. There should be one authoritative source that knows how to translate between colors and labels (from a computational perspective -- subsequent label visualization can use arbitrary colorization schema.) As it is, the server can use any rgba value of the form (255, 255, b, a) and for all valid b and a, RenderEngine will interpret that to be "empty".
(3) The idea that the definition of "empty" (or any of the other special label values) must persist across possible changes to label bitwidth doesn't seem reasonable. This seems to suggest that it is the computation color version of the label images that get persisted and would need to be interpreted down the road. It seems more likely that the actual label images is what would actually get persisted.
(4) Background color should be passed in the RPC request along with some of the other meta data included in addition to the glTF data. No server should be hard-coding any values like that.

From reading through these, my take is emitting the special label RGBAs (or a subset of it) and documenting them can be the common ground at the risk of being verbose for the protocol?

zachfang, 2022-12-26:
Another discussion regarding special labels.

Zach:
> For (4), there is a potential use case for RenderLabel::kDontCare as well for fixed environmental objects, e.g., walls or homecart. Thus designated colors for the special labels might be useful.

&

Sean:
> It's worth noting, all other labels get assigned directly to geometries so it's not necessary to pass the color values; they are properly anchored to something that was explicitly communicated. The background color is unique because there is no background "object".
Unless you're suggesting the renderer itself could generate artifacts that it wasn't told to generate which might otherwise be visible in a label image? Even as I write those words, everything about that seems wrong and terrifying.

The use case I was referring to is we might have some fixture-like geometries from the server side but not the client's glTF, e.g., the wall or the floor. We might not want to send everything through the wire to increase throughput. Those geometries will appear in the rendered images (color or depth), and I think `kDontCare` label is the most suitable in this case (compared to creating new labels on the server side).

jwnimmer-tri, 2023-01-03:
> From reading through these, my take is emitting the special label RGBAs (or a subset of it) and documenting them can be the common ground at the risk of being verbose for the protocol?

I can live with transmitting the background RGBA in the RPC request.  I still think specifying either black or white in the documentation is easiest to specify and implement and has no meaningful downsides, but it's basically a wash.

> The use case I was referring to is we might have some fixture-like geometries from the server side but not the client's glTF, e.g., the wall or the floor.

To my intuition, anything not transmitted by the client (e.g., walls or floor known only to the server) should be "background".

zachfang, 2023-01-26:
Tagging @SeanCurtis-TRI for further discussions. I plan to work on this next week once we agree on a solution. 

> anything not transmitted by the client (e.g., walls or floor known only to the server) should be "background".

My initial thought is the fixtures will have a general label as they are still some kind of object in contrast to the empty background. However, applying the empty label to them works for me as well.

jwnimmer-tri, 2023-05-08:
From f2f: we are happy to map both empty (i.e., "sky") and "random blender stuff like homecart frame" both to `RenderLabel::kDontCare`.

jwnimmer-tri, 2023-05-08:
We also seem to have come to consensus of specifying the RPC as "unknown labels are white" in its documentation, and not changing the message protocol at all.

zachfang, 2023-05-11:
To clarify, the "background" or any geometry added via any glTF server will be mapped to `RenderLabel::kDontCare`. However, we still treat "background" as `RenderLabel::kEmpty` in RenderEngine{Vtk, Gl}. That's the discrepancy we settle for now across different RenderEngne implementations (or GltfClient vs. the other two).

We have a test to compare output from VTK to the VTK-glTF-server, and I am trying to fix that.

jwnimmer-tri, 2023-05-11:
The information we need to communicate goes like this:

For type=label renders:
(1) The server must to turn off all spot lighting, shadows, anti-aliasing, etc. so that the diffuse rgb colors come back exactly.
(2) For geometry that did not come from the RPC (e.g., background image, blend file, etc.), the server must always texture it as white (with 100% alpha).

The is the only information we need for the RPC contract.

If we like, we can document in MakeRenderEngineGltfClient class overview that the only "special" label returned will be kDontCare.

*The server RPC does not know about labels.  Any discussion of RenderLabel in the RPC docs is a layering-violation defect.*

zachfang, 2023-05-11:
Okay, what you mentioned on the server side is consistent with my current change in https://github.com/RobotLocomotion/drake/pull/19397/files.

My main question is on the client side. Given the same SDFormat file to set up a scene, RenderEngineVtk will come back with `kEmpty` background, but RenderEngineGltfClient (regardless of the server backend) will have a background value of `kDontCare`. If that's confirmed, I can fix the integration test in RenderEngineGltfClient which compares output between VTK and GlrtfClient.

jwnimmer-tri, 2023-05-11:
Sounds fine.

closed 2023-05-16 by SeanCurtis-TRI

zachfang, 2023-05-16:
To summarize, **the background and any server-only geometries will be returned as white in the colored-label image for all the render server implementations**.

RenderEngineGltfClient will then convert that special pixel value to `RenderLabel::kDontCare`. By that, we admit the background label of RenderEngine{Vtk, Gl} will be different from RenderEngineGltf.

PR #19397 (2023-05-16): [geometry] Map white color to kDontCare in render_gltf_client
Close #18311 
Close #18425

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19397)