RobotLocomotion/drake #18314: UnitInertia::SolidMesh(const TriangleSurfaceMesh&)

RussTedrake, 2022-11-12:
Related to #17267, and probably required to complete that issue, I need a method to compute the unit inertia for the enclosed volume of a triangle surface mesh.  I believe that the algorithm is a small variation on the one we are using for computing the volume / centroid: https://github.com/RobotLocomotion/drake/blob/f7d728144d4d350fc5fa491a5182d02782bdacbd/geometry/proximity/make_convex_mesh.cc#L51 

Also related to #18312.

Once this exists, we should use it in `detail_mujoco_parser` to properly compute the inertia for imported meshes.

DamrongGuoy, 2022-11-15:
I confirm that this feature request is to add this function (similar to UnitInertia::SolidSphere()) :
```C++
namespace drake {
namespace multibody {
template <typename T>
class UnitInertia : public RotationalInertia<T> {
 public:
  /// Computes the unit inertia for a unit-mass solid mesh of uniform density.
  /// taken about its centroid.
  static UnitInertia<T> SolidMesh(const TriangleSurfaceMesh<T>& mesh); 
};
}  // namespace multibody
}  // namespace drake 
```

DamrongGuoy, 2022-11-15:
@SeanCurtis-TRI, I vaguely remember you mentioned something related to this before, right?

SeanCurtis-TRI, 2022-11-15:
Generally, it is closely related. 

I believe, we should actually be returning some variation of *three* quantities: mass, center of mass, and inertia tensor (with either a given density or an assumed density equal to one).

FCL has this [implemented for their own convex shape](https://github.com/flexible-collision-library/fcl/blob/master/include/fcl/geometry/shape/convex-inl.h#L100-L158) (which should be easily adapted to our triangle surface mesh representation).  It cites such sources as [geometrictools](https://www.geometrictools.com/Documentation/PolyhedralMassProperties.pdf).  Although, we'd want to merge the computations of volume (total mass), center of mass, and inertia tensor into one algorithm because of the ridiculous number of shared computations.

SeanCurtis-TRI, 2022-11-15:
I've got a few free cycles and I'm going to put a quick PR together.

SeanCurtis-TRI, 2022-11-18:
I didn't go the `UnitInertia::SolidMesh` route in #18345. Given that, in practice, you need mass, com, *and* unit inertia *anyways*, I figured it made more sense to go *straight* to the `SpatialInertia`. Also, as long as we're headed towards "just give me a mesh, I'll figure out mass", it made just as much sense to replace "mesh" with "box", "sphere", "ellipsoid", etc.

SeanCurtis-TRI, 2022-12-05:
#18345 adds a single entry point to get mass, center of mass, and unit inertia (i.e., a `SpatialInertia`) all in one blow. That should be sufficient go from geometry --> body.

closed 2022-12-05 by SeanCurtis-TRI

rpoyner-tri, 2023-05-03:
https://github.com/RobotLocomotion/drake/pull/19342 actually tries to close the loop from what Russ' mujoco implementation wants to what Sean's inertia algorithm provides. PTAL.