RobotLocomotion/drake #18323: [pydrake] "meldis -w &" did not open Meshcat window

DamrongGuoy, 2022-11-15:
### What happened?

Follow the slack discussion https://drakedevelopers.slack.com/archives/C43KX47A9/p1668457577658899, this command does not open Meshcat window:

```
bazel run //tools:meldis -- -w &
```
as suggested by meldis.py's [help message](https://github.com/RobotLocomotion/drake/blob/b87916821078da7f6562d4bff4049ddc47058021/bindings/pydrake/visualization/meldis.py#L21).

As shown in the `Relevant log output` below, the process is stopped like this:
```
[1]+  Stopped                 bazel run //tools:meldis -- -w
```
If I did `fg %1`, it will continue to open Meshcat window successfully.

Using `git bisect`, the problem started from [PR 18180](https://reviewable.io/reviews/robotlocomotion/drake/18180).

### Version

f63f19a76836354f394b7d8e96bb4395f68f70a9

### What operating system are you using?

Ubuntu 20.04

### What installation option are you using?

compiled from source code using Bazel

### Relevant log output

```shell
damrongguoy@TRI-18388202:~/GitHub/DamrongGuoy2/drake (master)$ bazel run //tools:meldis -- -w &
[1] 205094
INFO: Analyzed target //tools:meldis (0 packages loaded, 0 targets configured).
INFO: Found 1 target...
Target //bindings/pydrake/visualization:meldis up-to-date:
  bazel-bin/bindings/pydrake/visualization/meldis
INFO: Elapsed time: 0.548s, Critical Path: 0.00s
INFO: 1 process: 1 internal.
INFO: Build completed successfully, 1 total action
INFO: Build completed successfully, 1 total action

[1]+  Stopped                 bazel run //tools:meldis -- -w
damrongguoy@TRI-18388202:~/GitHub/DamrongGuoy2/drake (master)$
```

trowell-tri, 2022-11-15:
The "stopped" condition suggests `meldis` is trying to read from the terminal, but that PR has no changes to `meldis` to make it do that. That PR does have changes to the `_visualization_extra.py` which will change what's imported, however.

Commenting out the `ModelVisualizer` import in `_visualization_extra.py` and in `visualization_py.cc` cures the issue but I'm no closer to an answer than that, yet.

jwnimmer-tri, 2022-11-15:
The description so far sounds like one of these imports is likely to be the source of the problem:

https://github.com/RobotLocomotion/drake/blob/15e90cbda7aeb2497ea32144e743b668043f3d6e/bindings/pydrake/visualization/_model_visualizer.py#L1-L27

trowell-tri, 2022-11-15:
`PlanarSceneGraphVisualizer` appears to be implicated (if ModelVisualizer doesn't import that, then Meldis is happy, anyway). Meldis doesn't seem to use that function and it doesn't appear to be related to a double import there.

trowell-tri, 2022-11-15:
@jwnimmer-tri it seems to be `ConnectPlanarSceneGraphVisualizer` but no idea why yet.

trowell-tri, 2022-11-15:
(Also I'm curious what path led you to those imports so I can learn)

jwnimmer-tri, 2022-11-15:
I would try the same "comment out imports to see which one is at fault" for the [planar_scenegraph_visualizer imports](https://github.com/RobotLocomotion/drake/blob/15e90cbda7aeb2497ea32144e743b668043f3d6e/bindings/pydrake/systems/planar_scenegraph_visualizer.py#L1-L27).

> (Also I'm curious what path led you to those imports so I can learn)

If the only change related to meldis in the blamed PR was importing another module, but not calling any code from that module, then something in that module's imports must be the cause, since that was the only new code being "run" in the first place.

jwnimmer-tri, 2022-11-15:
To help bias your search priors: it would not surprise me if the `matplotlib` imports were the involved party.  `matplotlib` does a _lot_ of weird initialization logic during import, and I wouldn't be surprised if some of it tried to tickle stdin.

trowell-tri, 2022-11-15:
OK, so similar to the path I took, but more concisely. I'm scanning the planar scene graph and pyplot visualizer code quickly to see if I can see a likely culprit, else I'll do the code bisect there that I did in model_visualizer. But what confuses me is why this same condition doesn't STOP a model_visualizer run in the background.

trowell-tri, 2022-11-15:
> To help bias your search priors: it would not surprise me if the `matplotlib` imports were the involved party. `matplotlib` does a _lot_ of weird initialization logic during import, and I wouldn't be surprised if some of it tried to tickle stdin.

I almost mentioned `matplotlib` as a confounding factor in a prior comment. I've been deep in that code and it is a … thing.

trowell-tri, 2022-11-15:
FYI looks like `matplotlib.animation`

jwnimmer-tri, 2022-11-15:
I think it would be fair to move the `import matplotlib.animation as animation` from module-scope to the (two) functions where it's used.  That module _forks subprocesses_ while being imported (to check for Ubuntu 12.04 being broken) so is not a great candidate for imports up-front, especially when it will go unused in most cases.

jwnimmer-tri, 2022-11-15:
Indeed, running meldis with `&` to get it wedged, the process tree clearly shows `ffmpeg` as a child process.  That's consistent with this import-time code (from `animation.py`) being run and getting stuck:

```py
            # Ubuntu 12.04 ships a broken ffmpeg binary which we shouldn't use.
            # NOTE: when removed, remove the same method in AVConvBase.
            and b'LibAv' not in subprocess.run(
                [cls.bin_path()], creationflags=subprocess_creation_flags,
                stdout=subprocess.DEVNULL, stderr=subprocess.PIPE).stderr)
```

trowell-tri, 2022-11-15:
I was in the midst of doing the `animation` import in the `__init__` but your fix was faster. It fixes the issue, and also allows the `--pyplot` option in model_visualizer to still work. But the import in `__init__` seems to be in aid of handling some other edge cases without tkinter support. I'll submit this patch.

trowell-tri, 2022-11-15:
BTW I didn't notice that particular stanza when I examined `animation.py`, which I did just on their github repo as a first pass. We must be running an older version of matplotlib (I didn't check, again as a first pass). Perhaps a new matplotlib would also fix this?

jwnimmer-tri, 2022-11-15:
I don't think an acceptance test with background processes would be helpful.  (It would be difficult to write.)

I think the worthwhile test would be amend an existing test which imports all of pydrake (probably `all_each_import_test`) and add a test condition there that `sys.modules` does not contain any of our disallowed modules (`scipy`, `matplotlib.pyplot`, `matplotlib.animation`, `pydot`, `PIL`, `cv2`) -- things that are specifically imported within functions instead of a the module level.  The test should show a failure when the #18324 fix is absent.

trowell-tri, 2022-11-15:
That sounds like a worthwhile one to write. Want me to handle that one?

jwnimmer-tri, 2022-11-18:
Yes, please.

PR #18324 (2022-11-15): defer import of matplotlib.animation in pyplot_visualizer
In some deeply nested import situations, importing matplotlib.animation will hang a backgrounded Python program because it's reading from stderr. Deferring the import solves the issue. Resolves #18323.

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/18324)