RobotLocomotion/drake #18327: Add support for macOS Ventura

svenevs, 2022-11-15:
- [ ] Identify the locations drake warns about unsupported OS / compiler (only a few).
- [ ] x86 vSphere images can be built at any time, noting that we should select Xcode 14.1 for Ventura.
- [ ] arm64 orka CI is on hold until ~early December 2022 (#18287)

@jwnimmer-tri please let us know how to proceed / priority of this ticket (specifically, if we want to do them separately or keep x86 on hold until arm64 is available).

CC @BetsyMcPhail 

NOTE: a full local build on Ventura on an arm64 macmini (tri-macmini-01) using Xcode 14.1 (Monterey CI is currently 14.0.1) indicated that all of drake can compile.  Tests need to be rerun there (gurobi license server connection issues), more results to be posted here.

jwnimmer-tri, 2023-01-12:
The first goal should be x86 only.  It's fine to defer arm64 support until we have Orka remote caching working.

jwnimmer-tri, 2023-02-02:
Update from meeting: because we ship releases on Monterey, we can err on the side of most CI jobs still being on on Monterey.  We just want a few Ventura jobs to sanity-check that all tests pass.  Possibly just 1 bazel release + 1 bazel everything release would suffice.

jwnimmer-tri, 2023-02-22:
Also once we make progress here, we should fix https://drake.mit.edu/installation.html and https://drake.mit.edu/from_source.html to mention Ventura.  (It's somewhat embarrassing that we can't claim that yet.)

jwnimmer-tri, 2023-02-23:
> The first goal should be x86 only.

I take it back.  Let's not do anything on vSphere.

If we can get even an arm release build or few going, that's a great start.

svenevs, 2023-05-04:
https://drake-jenkins.csail.mit.edu/view/Mac%20Ventura/

We have ventura arm64 support.  13.3.1, xcode 14.3 is what is running in CI.

jwnimmer-tri, 2023-05-04:
Known TODO:
- The os.bzl is spewing a warning (b/c there was no CI); now that there is CI, remove the warning.
- Update docs stating support for Ventura.
- A few build error(s) with new Xcode.

svenevs, 2023-05-04:
@jwnimmer-tri I updated the checklist after scraping the log, I think I got most of them.

1. For the `std::move` thing, I tried to add `--copt='-Werror' --cxxopt='-Werror'` to get it to crash but that doesn't appear right.  Is there an easy way to force `-Werror`?  I see `-Werror` in other places, but can't trigger error for this warning specifically.
2. How much do we care about warnings coming from externals.  We've got `move` vs `std::move` warning, `curl` using unsafe `sprintf`, and a lot of `warning: a function declaration without a prototype is deprecated in all versions of C and is not supported in C2x [-Wdeprecated-non-prototype]` from C deps going on.
3. Once the easier fish have been fried and the build logs become more manageable, most of the test failures are related to solver tolerances so it would probably be best to punt back to TRI after we can get things cleaned up a bit first.

jwnimmer-tri, 2023-05-04:
(1) See [drake_cc.bzl](https://github.com/RobotLocomotion/drake/blob/master/tools/skylark/drake_cc.bzl), probably the `APPLECLANG_VERSION_SPECIFIC_FLAGS` will be the right place to say `-Werror=unqualified-std-cast-call`.

Separate from PR'ing that Werror change, we anyway should PR the code fixes to remove `using std::move;` throughout Drake.  I could do that, if it helps?

(2) Generally externals should mute all warnings, so `copts = ["-w"],` in their BUILD rule(s).

svenevs, 2023-05-04:
Ok thank you. With an error flag it's easy for me to whittle them down, but if you would rather do it I certainly won't stop you.  I'd like to fix that before asking people to look at solvers issues since currently the build log is really cluttered.

I'll be able to do that tomorrow.

jwnimmer-tri, 2023-05-04:
I'll do the `std::move` today, then.  I don't need the error flag, it's really just perl pie to delete `using std::move;`.  Every single time that line appears it's wrong, and should be deleted.  That *should* get all of the warnings gone.  I guess anything that's remaining you can follow-up on tomorrow.

jwnimmer-tri, 2023-05-04:
All but two files worth of `std::move` is fixed.  In your PR that adds the `-Werror...` flag for it, we can fix those remaining two files.

As of the PRs shown in the timeline above, I think all of the build **errors** are fixed.

I haven't looked at the tests yet.  For tolerancing failures, general procedure would be to open one PR per test that bumps the tolerance up so that it passes, and then assign the PR feature review the test-owner.  They can either LGTM it, or push whatever other fix they prefer instead.

jwnimmer-tri, 2023-05-04:
As far as warnings -- we can move CI into Production (and advertise official support in our docs) even with warnings.  We can file follow-up tickets to stomp out any specific remaining warnings that seem worth fighting.

svenevs, 2023-05-07:
- Tolerance PRs are up and assigned, except for `//solvers:create_constraint_test` (first post edited)
- Warning against unsafe std call ended up being a bit more involved, #19380
- Remaining: vtk test

jwnimmer-tri, 2023-05-15:
All but two tests are passing now:

(1) The VTK rendering failure.  I don't plan on investigating this one myself.  I'll most likely disable it on Ventura+Arm and leave it ticketed as a follow-up.

(2) A solvers constraint one.  It's a tolerancing problem but the solution isn't yet obvious to me.  I'll keep investigating.

jwnimmer-tri, 2023-05-17:
The last two PRs are filed.  Once they merge, I think we should be ready to move into Production.

BetsyMcPhail, 2023-05-18:
It turns out that caching is already turned on for release jobs with existing logic in drake-ci scripts (https://github.com/RobotLocomotion/drake-ci/blob/59c36cca473de1a155c7be1aa2b18394f79965c5/driver/environment.cmake#L140).

The debug jobs caching can be sorted out as part of #19341

BetsyMcPhail, 2023-05-22:
Ventura jobs have been moved to production.

jwnimmer-tri, 2023-05-22:
This is done now, with a non-critical few things filed as follow-ups:
- https://github.com/RobotLocomotion/drake/issues/19424
- https://github.com/RobotLocomotion/drake/pull/19380

If there are more things still remaining, please open new issues.

closed 2023-05-22 by jwnimmer-tri