RobotLocomotion/drake #18334: Auto-reload in ModelVisualizer

jwnimmer-tri, 2022-11-16:
I common use case for ModelVisualizer is creating a new model from scratch.

When doing that, the user will typically
- edit their model file(s) in their IDE, then
- re-launch the ModelVisualizer to see the effect, then
- close the visualizer, then repeat.

It would help speed up their workflow if they didn't need to re-launch the visualizer every time.  Ideally, it would monitor the input file(s) for changes, and if there was a change either (1) automatically reload the model, or (2) provide a "reload" button for the user to click at their leisure.  If we do (2), maybe the button could always be there (i.e., without trying to detect changes).

SeanCurtis-TRI, 2022-11-17:
With the presence of the button, it would be *nice* if the visualizer reported that there *was* a change or not. Updating files in the wrong directory is a common enough mistake, it would be nice to help users realize that they don't see any changes because the visualizer says there *aren't* any changes.

trowell-tri, 2022-11-17:
That's a good idea. We should have some kind of error reporting anyway, for malformed model files (maybe it already does this, I haven't tried) so this is just an extension of that.

jwnimmer-tri, 2022-11-17:
At the moment, there is no way to ask the Parser "which files did you load?".  (Recall that files can include other files or meshes, so monitoring only the one filename we pass to the parser is _not_ sufficient on its own.)  We'd need to add that feature before ModelVisualizer could report whether or not anything changed.

trowell-tri, 2022-12-21:
I've added an optional (for now, via a command-line switch) reload button so we can try this out and discuss future directions.

jwnimmer-tri, 2023-01-24:
We discussed this a bit today.

My decision is that for now, we'll de-scope "auto" as the feature.  I think the "Reload" button is very useful on its own, and should suit this use case OK.

Trying to instrument "auto" would mean that the Parser needs to reach down into the SceneGraph to understand which exact mesh / material / texture files it loaded.  That level of implementation cost doesn't seem worth the benefit of the "auto" feature.

closed 2023-01-24 by jwnimmer-tri