RobotLocomotion/drake #18338: SapSolver doesn't work with PlanarJoint

RussTedrake, 2022-11-17:
### What happened?

I tried using Sap in the box-flipup gym env (my version of the one Grant has in https://github.com/RobotLocomotion/drake/pull/18178 ), and got the following error message on my screen:
```
RuntimeError: Limits for joints with more than one degree of freedom are not supported. You are getting this exception because a new joint type must have been introduced. SapDriver::AddLimitConstraints() must be updated to support this feature. Please file an issue at https://github.com/RobotLocomotion/drake.
```
I am pretty sure that this is because of 
```
    box_joint = plant.AddJoint(
        PlanarJoint("box", planar_joint_frame,
                    plant.GetFrameByName("box", box_instance)))
```

### Version

master

### What operating system are you using?

macOS 12 (Monterey)

### What installation option are you using?

compiled from source code using Bazel

sherm1, 2022-11-29:
Have you seen this, @amcastro-tri ?

xuchenhan-tri, 2023-01-18:
Looks like the SapDriver mistakenly determined that the planer joint has limits imposed. I can take this one.

xuchenhan-tri, 2023-01-18:
Looking at the referenced branch from @RussTedrake, it does seem like joint limits are added for the planar joint in question. In that case, the throw message stands correct for the moment. I'll change the label to reflect this is a feature request.

xuchenhan-tri, 2023-01-18:
The SapLimitConstraint class can be extended to support PlanarJoint and similar multi-dof joints where q̇ = v. We would probably need the joint to self-report whether q̇ = v is true. Other multi-dof joints are trickier.