RobotLocomotion/drake #18346: //systems/lcm:lcm_interface_system_test may have false positives

liangfok, 2022-11-18:
### What happened?

This test failed in the following nightly-production build:

https://drake-cdash.csail.mit.edu/test/898569555

It involves the follwoing code: https://github.com/RobotLocomotion/drake/blob/4c4b9f7c829f21822c57267f01916851bc7d15c9/systems/lcm/test/lcm_interface_system_test.cc#L108-L123

It appears there is a race condition between the transmit thread publishing and the time `simulator.reset()` and `HandleSubscriptions(0)` are executed, meaning it is possible for subsequent calls to `drake_lcm->HandleSubscriptions(1)` to still return a non-zero value.

### Relevant log output

```shell
[ RUN      ] test/LcmInterfaceSystemTest.AcceptanceTest/3
systems/lcm/test/lcm_interface_system_test.cc:123: Failure
Expected equality of these values:
  total
    Which is: 1
  0
[  FAILED  ] test/LcmInterfaceSystemTest.AcceptanceTest/3, where GetParam() = 3 (448 ms)
```

jwnimmer-tri, 2022-11-20:
Closing per "The first time an unexplained failure occurs, close the issue immediately..."
-- https://drake.mit.edu/buildcop.html#process

ggould-tri, 2023-01-04:
https://drake-jenkins.csail.mit.edu/view/Production/job/mac-arm-monterey-clang-bazel-nightly-everything-release/79/
Reopening per procedure; this is a twice in three months bug so lowest priority.  Assigning @sammy-tri (LCM owner) for delegation and triage.

jwnimmer-tri, 2023-01-04:
We should just `flaky = True` in the BUILD file.

PR #18561 (2023-01-09): Mark lcm_interface_system_test as Flaky
Closes #18346

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/18561)
closed 2023-01-09 by SeanCurtis-TRI