RobotLocomotion/drake #18348: ModelVisualizer says "Use Ctrl-C or..." even when running in jupyter

RussTedrake, 2022-11-19:
### What happened?

I love the upgraded `ModelVisualizer`  But I noticed that even running in a notebook (like the authoring multibody simulations tutorial), calling `visualizer.Run` prints to the screen
```
INFO:drake:Meshcat listening for connections at http://localhost:7000
Use Ctrl-C or click 'Stop Running' to quit
```
That's not true in the jupyter notebook workflow.  Pressing Ctrl-C doesn't work.  (And I would have liked to think that the keycode "Escape" would have worked if I pressed Escape in meshcat.

trowell-tri, 2022-11-28:
It's the "Use Ctrl-C" part of the message that's the issue, right? The old tutorials printed the same message (but without the "or click 'Stop Running'" part) and so while that bothered me I figured it was something we were living with. It would be great to do better and it looks like there are some janky ways to tell if Python is running inside a notebook but is there a way that we like to use?

Great point about allowing Escape to quit as well, I'll take a look at that.

RussTedrake, 2022-11-28:
Correct.  "Ctrl-C" is the issue.  It does not have the desired effect in a jupyter notebook.  I think matching the (JointSliders behavior)[https://github.com/RobotLocomotion/drake/blob/a67b41e51550aec7a55c3ad649f213534ba2bc58/multibody/meshcat/joint_sliders.cc#L267] would be reasonable, whether we are in jupyter or not.  (A python user presumably knows they can press Ctrl-C also, we don't need to tell them?)

PR #18411 (2022-12-07): Allow pressing Escape to stop ModelVisualizer
Also change the message that ModelVisualizer prints so that it is accurate when running in a notebook, under the theory that someone running the script from the command line will understand that they can also press Ctrl-C to quit. This brings this script more into line with JointSliders::Run.

Fixes #18348

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/18411)
closed 2022-12-07 by trowell-tri