RobotLocomotion/drake #18350: [multibody/parsing] Parsers cannot override default collision filters set in Finalize

RussTedrake, 2022-11-21:
`MultibodyPlant::Finalize` calls `ApplyDefaultCollisionFilters`, which will, for instance, demand that parent and child geometries are filtered.  I needed to overload that capability in order to correctly parse a mujoco file I was working on, but now I don't believe it's possible.  One cannot parse after `Finalize`, so any changes at the parser level are getting blown away.

Looking at the sdf parser tests, it confirmed my fears.  All of those tests are done pre-finalize.
```
  // Test collision filtering across models. Done before Finalize() to exclude
  // default filters.
```

cc @SeanCurtis-TRI .

rpoyner-tri, 2022-11-28:
I wrote that comment, and I don't believe that is what it means.What I meant was more like "building an expected value that re-calculates welded subgraphs makes for a cluttered and confusing test. Instead, do simple tests before the Finalize() step has added more filters to the model."

Now, let us consider the hypothesis of the issue title. Certainly, parsers run before Finalize(), so (currently), they can't override *any* modifications done at Finalize(). So, then, the question becomes: do the current default collision filters *need* overriding? 

The current implementation of ApplyDefaultCollisionFilters (ADCF) implements two rules: (1) topologically adjacent body pairs are filtered, (2) all pairs connected through a wellded subgraph are filtered. I'm no roboticist, but those rules seem pretty reasonable for a graph of rigid bodies. Let's see where they came from.

Rule 1 has been in place for a long time -- PR #8884 in June of 2018. I'm having a little trouble tracing the history of why the rule was implemented, but it seems intuitive, at least.

Rule 2 has been in place for less than a year -- PR #16920, in direct response to issue #16422. The rule is a generalization of the older rule "filter all bodies welded to the world body". The issue and motivating slack discussion show attempts to balance user convenience (don't make me hand-write collision filters) with conservatism (don't filter out pairs that could collide and be of interest).

Does your use case need to rewrite the filters implemented by the current rules? Re-reading your comment, it sounds like you need modifications to Rule 1. I could imagine that happening if the adjacent bodies have complicated geometries. I wasn't around for the implementation of Rule 1, but I've heard  justifications like the following. Some models use simple geometries (box, cylinder) where normal joint motion produces "collisions" because the geometry is too approximate to keep the joint motion region clear.

RussTedrake, 2022-11-28:
You are correct.  The issue was Rule #1.  I think it is a very reasonable default, and the correct answer in most cases.  But I also think that one should be able to override it if they understand the consequences.

rpoyner-tri, 2022-11-28:
Ah,ok. Now we have the design question of how to satisfy the minority, while not messing up the maybe buried assumptions of existing programs. While collision filters can be changed both before and after Finalize(), as far as I know the only notion of priority is most-recent-change wins. So that says to me that we probably have to do some work to get the desired result. Here are some possibilities. I'm sure there are more I've missed.

 * Make Rule 1 more conservative in some way that "senses" cases where it is not appropriate (not a breaking change, maybe, but spooky)
 * Add some API and/or model file directives that disable some or all default filtering (not breaking, but maybe awkward)
 * Remove default filtering, and provide separate APIs/directives that apply the rules (breaking, and potentially awkward)
 
\cc @SeanCurtis-TRI @jwnimmer-tri for design thoughts.

sherm1, 2022-11-28:
FWIW I like the second option (something you can put in the sdf file that affects the later behavior of ApplyDefaultCollisionFilters()).

Note that it is possible to have multiple geometries attached to a single rigid body. The purpose of "rule 1" is to allow clunky collision geometry around joints w/o jamming. A user might want other geometries on the same bodies to collide. Depending on whether that's a meaningful use case, it might make sense for the sdf signal to operate on geometries rather than (or in addition to) whole bodies.

rpoyner-tri, 2022-11-28:
Maybe what we are struggling with here are the differences between drake's methods of dealing with collision pairs/filtering and mujoco's. Looking at the mujoco `option/collision` [mode setting,](https://mujoco.readthedocs.io/en/stable/XMLreference.html#option) it seems like the mujoco semantics are very different. 

One could imagine using the `option/collision` field to signal some different handling of collision filters. However, drake is pretty promiscuous in terms of allowing mixed format model files. What are the semantics in a plant provisioned by model files of different formats? Does it then vary by model instance?

rpoyner-tri, 2022-11-28:
FWIW, I haven't even gotten to the point with mujoco's collision modes to say which is most like drake, or whether any of the existing modes could be considered "supported".

RussTedrake, 2022-11-29:
The issue came up when I was working on the Mujoco parser, but it isn't specific to Mujoco.  The possibility of wanting to include collisions between parent and child is a valid request for any file format.

If we agree on the issue, I'm ok if we mark it low priority.  I don't know if it's the most pressing, but it seemed a basic flaw in our parsing pipeline (again not mujoco specific).

rpoyner-tri, 2022-11-29:
That's fair. I do think we need to have a good transition plan, since implicit dependencies on "rule 1" are probably buried far and wide.

jwnimmer-tri, 2023-01-03:
To be clear, Rico outlined two heuristics (Rule 1 and Rule 2).  I think we all agree that Rule 2 is fine to leave alone entirely for now.

For Rule 1, I'll propose this:

(a) Add config option to MbP for whether or not this rule shall be applied during Finalize.  Default it to the current semantics.

(b) When a parser (e.g. our Mujoco parser) encounters an input file that is incompatible with the in-effect MbP setting, that is a parse error.  The error message tells the user how to reconfigure their MbP to work around it.

That provides for a way to load Mujoco files successfully, as well as for SDFormat etc users to disable the filtering for their own purposes as well.

RussTedrake, 2023-01-03:
That seems like a reasonable starting place to me.

rpoyner-tri, 2023-01-24:
@SeanCurtis-TRI notes that since the free-body/joint changes, Rule 1 may be even more wrong. Maybe not a concern for this issue ("optionally turn off rule 1"), but perhaps a new issue or PR to fix Rule 1 if necessary.

SeanCurtis-TRI, 2023-01-24:
I was looking at the [code](https://github.com/RobotLocomotion/drake/blob/master/multibody/plant/multibody_plant.cc#L1113-L1121). I think generally, we're ok because we explicitly state that if the parent of the joint is the world, we *don't* filter.

Furthermore, when #18390 added the `QuaternionFloatingJoint`, the "adjacent body collision filter" logic also excluded bodies connected by that joint. So, we're good.

jwnimmer-tri, 2023-01-30:
See also #5211.

rpoyner-tri, 2023-02-03:
#18736 implements Nimmer's (a) above. I have no idea how a parser would implement (b). How does a parser tell that the model file dislikes some not-yet-applied collision filters?

It's not a great fit either for (say) the Mujoco parser to just flip the bit, since the bit is plant-global, and a plant could be provisioned from any number of files in any combination of formats.

jwnimmer-tri, 2023-02-03:
My idea for (b) was that the Mujoco parser considers `plant.get_adjacent_bodies_collision_filters()` during the parse.  If it encounters a Mujoco input file where the value of that plant setting must be `false` in order to load the file correctly, then it raises a `diagnostic.Error` telling the user that the input file is not supported, unless and until they configure their plant correctly.

If the user wants to load _more_ than just the one Mujoco model (not likely!), but wants to have adjacent filters for their other SDFormat models, they get to add those filters by hand, since it's a plant-global config setting.

jwnimmer-tri, 2023-02-03:
> I needed to overload that capability in order to correctly parse a mujoco file I was working on,  ...

@RussTedrake Discussions for how do part (b) here would probably be much clearer if we had this particular Mujoco file posted into the ticket for reference.  Could you provide an example, or at least point to a part of the MJCF spec with the kind of data format options that would be relevant here?

RussTedrake, 2023-02-05:
The test in #18351 shows an [example of a trivial mujoco file that cannot be parsed properly](
https://github.com/RobotLocomotion/drake/blob/254805657decb614eb531932879003f674dc7942/multibody/parsing/test/detail_mujoco_parser_test.cc#L976).

The relevant mujoco documentation is [here](https://mujoco.readthedocs.io/en/latest/XMLreference.html#contact). The contact elements should allow models to opt-in to collisions, which currently is only meaningful for the models that where collisions are ignored by default.  In this case, we cannot parse a model specifying a <pair> that specifies that collisions should occur between a parent and a child.

SeanCurtis-TRI, 2023-02-06:
Ok. If I understood all of that, mujoco's collision filters are, by default, the same as SceneGraphs: no two geometries affixed to the same body are considered for contact. But all other pairs are. One can do the following:

1. Configure to use "predefined" which means only pairs of *geometries* explicitly listed get tested.
2. Configure to use "dynamic" which includes any *geometry* pair the broadphase can't cull.
3. One can *exclude* body pairs (i.e., no geometry from one body can collide with any geometry from the other body). This is the semantics that MbP has when it assigns its own collision filters based on topology (body-centric). However, the global setting *cannot* be set to "predefined" (see above).

So, this is why simply turning off MbP's creation of *any* collision filters at all is a reasonable coarse approximation of Mujoco semantics. It doesn't allow for collision pairs or excluded pairs. It is compatible with the global collision configuration being "dynamic" (with no exclusion pairs defined) or "all" (with no collision pairs and no exclusion pairs).

jwnimmer-tri, 2023-02-07:
I think #18736 closed this issue, didn't it?

closed 2023-02-08 by rpoyner-tri