RobotLocomotion/drake #18352: Upgrade Gurobi

hongkai-dai, 2022-11-21:
Gurobi 10 is released https://www.gurobi.com/whats-new-gurobi-10-0/

I will contact the gurobi vendor to check if we need to upgrade our licenses.

jwnimmer-tri, 2023-01-27:
@hongkai-dai do you have any timeline / goal for this upgrade?

hongkai-dai, 2023-01-27:
I will do it next week.

RussTedrake, 2023-06-22:
I guess we never finished this upgrade?

hongkai-dai, 2023-06-22:
Oops, sorry for the delay. I will work with Gurobi and TRI's IE team to upgrade our license server.

jwnimmer-tri, 2023-07-13:
Actually there are two things possibly required -- update the CI provisioning to install Gurobi 10, and then (if necessary) also upgrade the license.

hongkai-dai, 2023-07-13:
BTW, Gurobi just received our payments and updated our license to the newest version. We will need to run `grbgetkey` to obtain the new license.

Does the Drake CI uses TRI's gurobi license?

RussTedrake, 2023-07-13:
> Does the Drake CI uses TRI's gurobi license?

No.

BetsyMcPhail, 2023-07-24:
Using the prior upgrade issue (#16604) as reference, there are several steps that need to be coordinated to upgrade Gurobi:

- [ ]  Update required version in Drake itself
- [ ]  Update [Gurobi cask]
- [ ]  Update Drake CI scripts
- [ ] Check if current license will support 10.0 and upgrade if needed
- [ ]  Update license server. We may have a new license and a new software version, these can both be updated at the same time.
- [ ]  Update CI provisioned Mac and Linux images

svenevs, 2023-07-27:
@BetsyMcPhail lets coordinate offline, rather than needing a flag day transition in CI, we might want to stage it so that we can install two versions of gurobi.

https://github.com/RobotLocomotion/homebrew-director/tree/master/Casks

we can do versioning.  Lets talk offline when you are back about what will be the easiest long term solution.

BetsyMcPhail, 2023-08-02:
The CI License server has been updated (https://support.gurobi.com/hc/en-us/articles/360040451771-How-do-I-update-my-token-server-to-a-newer-version-).

Note that Gurobi licenses should be backwards compatible (https://support.gurobi.com/hc/en-us/articles/360013417471-How-do-I-request-a-license-for-an-older-version-of-Gurobi-#:~:text=Gurobi%20licenses%20are%20backwards%20compatible,C%3A%5Cgurobi%5C%20on%20Windows), so this step can be completed first, before any changes to drake / drake-ci / etc.

BetsyMcPhail, 2023-08-02:
Experimental job with both drake and drake-ci PRs: https://drake-jenkins.csail.mit.edu/view/Everything/job/linux-focal-unprovisioned-clang-bazel-experimental-everything-release/33

BetsyMcPhail, 2023-08-08:
Linux provisioned images are ready: **drake-linux-focal-2023-08-08** and **drake-linux-jammy-2023-08-08**

PR #19766 (2023-08-24): Upgrade to Gurobi 10.0.2
Resolves #18352

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19766)
closed 2023-08-24 by svenevs