RobotLocomotion/drake #18363: macOS arm jenkins jobs incomplete

svenevs, 2022-11-26:
We seem to be missing some macOS arm build flavors.

https://drake-jenkins.csail.mit.edu/view/Mac%20ARM/job/mac-arm-monterey-unprovisioned-clang-bazel-experimental-release/

There is no corresponding `debug` unprovisioned experimental build.  Probably a few more.  Flagging now before I forget.  CC @BetsyMcPhail

BetsyMcPhail, 2022-11-28:
There has historically never been any Mac unprovisioned debug builds. On VSphere, we would have to increase the unprovisioned image size. See #17869 for a discussion of the current matrix of Mac builds (specifically [here](https://github.com/RobotLocomotion/drake/issues/17869#issuecomment-1304591223))

svenevs, 2022-11-28:
I see, lets keep this one on hold then until after the orka updates.  I don't see why the vsphere limitations should prevent us from enabling unprovisioned debug experimental builds for arm64.  They currently have minute value to me, but I don't need CI for what I'm testing.

svenevs, 2024-01-11:
Closing as stale / irrelevant now.

closed 2024-01-11 by svenevs