RobotLocomotion/drake #18368: [doc] Release notes should be date- or "human"-sorted?

EricCousineau-TRI, 2022-11-28:
### What happened?

I glanced at [Release Notes](https://drake.mit.edu/release_notes/release_notes.html) pages, I thought the v1.10.0 release notes were missing.

I then realize it was because they were alphanumerically sorted:
> [image]

Possible solutions are to use date-based sorting, or use "human" sorting, e.g. in Python
https://nedbatchelder.com/blog/200712/human_sorting.html

\cc @rpoyner-tri (given that Jeremy's out)

### Version

master

### What operating system are you using?

Ubuntu 20.04

### What installation option are you using?

compiled from source code using Bazel

jwnimmer-tri, 2022-12-14:
Rough draft of the fix here:
  https://github.com/jwnimmer-tri/drake/commits/relnotes-sorting

\CC @rpoyner-tri or @xuchenhan-tri to carry over the finish line.  Jekyll sorts the index by the "date" by default, so we can use that to control the ordering.  We retain the "notes.released" metadata with the same information; trying to display "note.date" in list of notes displays the time (midnight) as well, which is not helpful.

rpoyner-tri, 2022-12-14:
https://github.com/robotlocomotion/drake/pull/18458

closed 2022-12-14 by rpoyner-tri