RobotLocomotion/drake #18370: [setup] Ubuntu Jammy may have libgcc-12-dev, which breaks mkdoc

EricCousineau-TRI, 2022-11-28:
### What happened?

See more details in https://drakedevelopers.slack.com/archives/C0KDGF4V9/p1669663654774429
\cc @calderpg-tri @ggould-tri @jwnimmer-tri 

As Grant indicated, can repro as follows:
``
bazel build @drake//bindings/pydrake:documentation_pybind.h
``

If you have `libgcc-12-dev` installed, but *not* `g++-12` (or some package between it and `libgcc-12-dev`), then it will fail with something like:
```
bazel-out/k8-opt/bin/external/drake/bindings/pydrake/documentation_pybind.h.tmp_artifacts/mkdoc_glue.h:1:10: fatal error: 'optional' file not found
Ubuntu clang version 12.0.1-19ubuntu3'
...
RuntimeError: The operating system's C++ standard library is not installed correctly.
```

If you run off of bare Ubuntu install (e.g. `ubuntu:22.04` in Docker / Apptainer), then 

### Version

2b49d9bf3a971620ec00673949d91339f8162c2c

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

EricCousineau-TRI, 2022-11-28:
Narrowed repro from missing `g++-12` to `libstdc++-12-dev`

jwnimmer-tri, 2022-11-29:
The solution path here will be to improve the error message from #15652 even more, perhaps by listing out some package names more specifically.  In general, our `install_prereqs` can't fix this automatically -- because we don't know what the user already has installed (and there's no way to change how `libclang` searches for GCC's libraries).

jwnimmer-tri, 2023-01-26:
Here's the tl;dr for how to fix this:

On Jammy if you install `libgcc-12-dev`, then in order to compile Drake you **must** also install `libstdc++-12-dev` (or uninstall `libgcc-12-dev`).

PR #18669 (2023-01-27): [setup] Automatically fix GCC 12 installation mistakes on Jammy
Closes #18370.

+@rpoyner-tri for both reviews, please.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/18669)
closed 2023-01-27 by jwnimmer-tri