RobotLocomotion/drake #18396: [geometry] AddContactMaterial sets double; GetHuntCrossleyDissipation retrieves T

EricCousineau-TRI, 2022-12-03:
### What happened?

In Anzu, I upgrading from Drake 2b49d9bf3a97 to abe2e88df2a9 (Anzu PR 9934), but encountered an error as described in title. I am not sure which commit is the offender.

In relevant code, there is a function, `MakePlant<T>`, meant for `T={double,AutoDiffXd}`.
(This is tested via Python bindings.)

`AddContactMaterial` will add `double`:
https://github.com/RobotLocomotion/drake/blob/abe2e88df2a9b3d55bdc6c0e689650cb956a67db/geometry/proximity_properties.cc#L38-L50

However, `internal::GetHuntCrossleyDissipation` expects `T`:
https://github.com/RobotLocomotion/drake/blob/abe2e88df2a9b3d55bdc6c0e689650cb956a67db/multibody/plant/contact_properties.cc#L59-L61

This causes hard-to-trace errors.

We should either make `AddContactMaterial` templated on `T`, or make `internal::GetHuntCrossleyDissipation` use `GetProperty<double>`.

Marking as high priority as this breaks Anzu, and seems like a fragile and inconsistent contract.

\cc @xuchenhan-tri @SeanCurtis-TRI 

### Version

abe2e88df2a9

### What operating system are you using?

Ubuntu 20.04, Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

EricCousineau-TRI, 2022-12-03:
From looking 2b49d9b...abe2e88, my guess is #18170 is what caused it?

EricCousineau-TRI, 2022-12-03:
#18397 is effectively a one-line functional change. I intend to land it to unblock updating Anzu, and will defer the more proper fix to the resolution of this issue (with lower priority).

EricCousineau-TRI, 2022-12-03:
I have merged #18397. Please note that I changed the types in there a few different ways, and all tests in Drake passed (meaning lack of test coverage).

One thing to consider for these type-erased properties (`ProximityProperties`) is scalar conversion.
From looking at `GeometryState`, I don't see anything that attempts to scalar convert the properties.

If we do not intend for that mechanism to exist, then we should stick with `double`.
If we want a more expressive `T` (`AutoDiff` or `Symbolic`), then we need to be diligent on how to express this via scalar conversion (e.g. parsing from URDF / SDFormat, then converting to `AutoDiff`, then adding gradients to these properties).

SeanCurtis-TRI, 2022-12-03:
This is not a straightforward solution.

Geometry properties live in the context but they do *not* get transmogrified because they live inside `AbstractValue`. So, if `MbP` goes to get `Foo<T>` but it was originally instantiated on `double`, it will fail to access its property. Currently, geometry properties should be considered to be constants and one cannot differentiate w.r.t. them.

I'd suggest that the `contact_properties.cc` is defective in misusing the previously existing API.

EricCousineau-TRI, 2022-12-04:
> I'd suggest that the `contact_properties.cc` is defective in misusing the previously existing API.

+1 to that in the sense of inconsistent workflows.
However, I still fear my merge of #18397 may disservice someone who was relying on this non-straightforward workflow.
I *think* it's possible to make the workflow "more straightforward".

> [...] but they do not get transmogrified because they live inside `AbstractValue` [...]

I don't think this is a hard blocker. Per above, two possible solutions for this:

1. Always require the value be of scalar type `T`. When transmogrifying, explicitly denote which properties must transmogrify, and explicitly loop through them and swap out the `AbstractValue` from `U` to `T`.
2. When using `T` where `T(double())` is valid (e.g. `AutoDiff`, `Expression`), then update call sites to retrieve `T` if possible, or `T(double_value)` otherwise. This 

I am in general a fan of (1) - I would love it if we got rid of most, if not all, notions of `VectorBase` being distinct from `AbstractValue`, and instead admitted a more fluid interchange of (still performant) functionality.

I am a fan of (2) in this specific case, because it seems like it'd allow for more workflows.

EricCousineau-TRI, 2022-12-04:
I think #18399 resolves this issue as it's currently stated.

There's still the issue of user expectations and flexibility when it comes to scalar conversion, as Sean and I have mentioned.
Should that be separate issue, or should this issue be rescoped to that?

EricCousineau-TRI, 2022-12-06:
Rescoped

EricCousineau-TRI, 2023-01-18:
Aye, will un-rescope it and mark as resolved by #18399.

closed 2023-01-18 by EricCousineau-TRI