RobotLocomotion/drake #18416: SpatialInertia factory functions for primitive shapes.

mitiguy, 2022-12-07:
Multiple PRs to implement SpatialInertia factory functions for primitive shapes.

- [x ] Create a "prototype PR" to provide a template for the function name, parameter names, documentation, and use in C++.
PR #18407 
- [ ] Create a PR that implements all the primitive shapes that are used in geometry_spatial_inertia.cc
- [ ] Create a PR that implements the remaining primitive shapes that are used in unit_inertia.cc
- [ ] Create a PR that implements a tetrahedron in unit_inertia.cc and spatial_inertia.cc.  Update geometry_spatial_inertia.cc
- [ ] Create Python bindings for all SpatialInertia factory methods for primitive shapes.
- [ ] Opportunistically, replace calls to the SpatialInertia constructor that uses density and a UnitInertia factory method for primitive shapes with the equivalent new SpatialInertia factory functions.

SeanCurtis-TRI, 2022-12-07:
Perhaps "Create a PR" is not what the checklist should be. It's not the creation of the PR that is of interest; it's the introduction of code into master. So, perhaps, it would be better to rephrase it and "uncheck" the current box.

mitiguy, 2024-01-10:
This issue was addressed in multiple PRs and now seems to be able to be closed.

closed 2024-01-10 by mitiguy