RobotLocomotion/drake #18435: Upgrade googlebenchmark  from 1.7.0 to 1.71

BetsyMcPhail, 2022-12-12:
Follow-up to #18379.

When attempting to update googlebenchmark, I get the following error:

```
~/develop/drake$ bazel-bin/tools/workspace/new_release --upgrade=googlebenchmark
Collecting bazel repository details...
INFO: All external dependencies fetched successfully.
Loading: 0 packages loaded
Upgrading googlebenchmark from v1.7.0 to v1.7.1
Downloading new archive...
Populating repository cache ...
ERROR: /home/local/KHQ/betsy.mcphail/.cache/bazel/_bazel_betsy.mcphail/4e753aaa1a8cee0cff3391e1459b0a22/external/googlebenchmark/BUILD.bazel:29:11: no such package '@libpfm//': The repository '@libpfm' could not be resolved: Repository '@libpfm' is not defined and referenced by '@googlebenchmark//:benchmark'
ERROR: Evaluation of query "deps(//...)" failed: errors were encountered while computing transitive closure
Loading: 8 packages loaded
Traceback (most recent call last):
  File "/home/local/KHQ/betsy.mcphail/develop/drake/bazel-bin/tools/workspace/new_release.runfiles/drake/tools/workspace/new_release.py", line 352, in <module>
    main()
  File "/home/local/KHQ/betsy.mcphail/develop/drake/bazel-bin/tools/workspace/new_release.runfiles/drake/tools/workspace/new_release.py", line 344, in main
    _do_upgrade(temp_dir, gh, args.upgrade, metadata)
  File "/home/local/KHQ/betsy.mcphail/develop/drake/bazel-bin/tools/workspace/new_release.runfiles/drake/tools/workspace/new_release.py", line 279, in _do_upgrade
    subprocess.check_call(["bazel", "fetch", "//...", f"--distdir={temp_dir}"])
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['bazel', 'fetch', '//...', '--distdir=/tmp/drake_new_release_uc1dfdub']' returned non-zero exit status 2.
```

I found a related google/benchmark issue: https://github.com/google/benchmark/issues/1510

rpoyner-tri, 2022-12-13:
I've looked into this, and have a working branch. It appears that I can just add libpfm4 as a drake dependency, and don't need to do any mac-specific magic, because of the way the google benchmark build works.

Tactical question: should I add libpfm4 in a commit by itself, then update benchmark in a separate commit? I suspect that flows more nicely for release notes.

rpoyner-tri, 2022-12-13:
See PR #18447.

PR #18457 (2023-01-09): [workspace] Upgrade googlebenchmark to v1.7.1
Closes #18435.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/18457)
closed 2023-01-09 by xuchenhan-tri