RobotLocomotion/drake #18445: RgbdSensor segfaults when there is no display

RussTedrake, 2022-12-13:
### What happened?

I've put a reproduction here:
https://deepnote.com/workspace/russtedrake-b58d8141-6691-43e8-ad76-0575b319c8b8/project/Camera-sim-wo-virtual-display-c4562e91-4005-44cd-9632-dd3fb3fbed96/%2Fcamera_sim.ipynb

We've known for some time that RgbdSensor crashes when there is no display (e.g. on a cloud machine).  We also know how to work around it with xvfb and pyvirtualdisplay.  I have an AddRgbdSensor method that tries to set up the virtualdisplay if needed: https://github.com/RussTedrake/manipulation/blob/3a71ff5ae086b5869cc2ee87b016bc196531c950/manipulation/scenarios.py#L232 .  But at least one student this week had their notebook crashing on every evaluation of the RgbdSensor output port.  Somehow their setup defeated my simple check, and the virtual display was not being set up.  I will try to get a reproduction for it.

My request in this issue: RgbdSensor should NOT segfault in this situation, but instead provide a useful error message.  The mysterious segfault caused lots of frustration and confusion.

Note: Unfortunately, running
```
(unset DISPLAY; bazel-bin/systems/sensors/rgbd_sensor_test)
```
was *not* enough to reproduce the error on my local machine.

### What operating system are you using?

Ubuntu 20.04

### What installation option are you using?

dockerhub

SeanCurtis-TRI, 2023-01-03:
@jwnimmer-tri @rpoyner-tri @ggould-tri @sammy-tri @EricCousineau-TRI 

Quick question for the people I think most likely to be able to answer this question:

What's the easiest way to reproduce this locally? How do I fool my execution context into *not* finding a display?

SeanCurtis-TRI, 2023-01-03:
@RussTedrake When I try running the deepnote, I end up getting a failure earlier than expected. I got this error while evaluating the `plot_camera_images()` block:

```
A kernel interruption error usually occurs for one of the following reasons:
 - The kernel process runs out of RAM. In this case, click here to upgrade your machine.
 - There is a bug in one of the libraries (e.g., version conflicts, missing binary dependency, etc)
```

I'm 99.9% deepnote ignorant and can easily believe I've done something wrong.

jwnimmer-tri, 2023-01-03:
> What's the easiest way to reproduce this locally? How do I fool my execution context into not finding a display?

I think running locally inside of Docker should _probably_ be enough to defeat the libGL display hunting.

That said, the codepath for errors when running on Deepnote without a display could easily be different than what happens in Docker.  If you develop in Docker, you should then also re-test in Deepnote.

jwnimmer-tri, 2023-01-04:
> I end up getting a failure earlier than expected.

I'm not sure where you expected an error, but that looks like where it would happen to my understanding.  The plot_camera_images evaluates the camera output ports, which should trigger rendering, which should segfault.  The "kernel interruption" is Deepnote's spelling of segfault.

RussTedrake, 2023-03-02:
It seems that we lost steam on this.  Deepnote is currently segfaulting for anyone who finds our rendering_mutlibody_plany tutorial (and presumably has been for some time).  @SeanCurtis-TRI -- have you been able to reproduce it locally?

RussTedrake, 2023-03-02:
FWIW -- asssuming we can avoid the segfault, a helpful error message would go a bit beyond "you need to set a display".  I don't think it would be unreasonable to write out something like:  "if you are running headless,  you might want to consider running xvfb." or even "on the Drake Deepnote tutorials, we run `/opt/drake/share/drake/setup/deepnote/install_xvfb` in the init.ipynb".  This would help people that copy the tutorial code/notebook, but not the project and miss the hidden `init.ipynb` and it's implications.

jwnimmer-tri, 2023-03-08:
Per f2f chat -- @RussTedrake might do a a near-term workaround here, which would be adding a cell to the top of the rendering notebook(s), where (iff running on deepnote) the cell would run the `/opt/drake/share/drake/setup/deepnote/install_xvfb` again as a subprocess.

This would work around the case where Deepnote somehow did not run the init notebook.

It would also make it more clear for users who copy only the _notebook_ instead of the _project_ for themselves, how to prepare the xvfb for rendering.

Even after that improvement, we should still keep this issue open -- with the goal of providing a good diagnostic in when VTK cannot access a display.

liruiw, 2023-04-23:
I recently came across another use case for headless rendering where I need to run drake in cluster without a screen and sudo access (wasn't able to use xvfb). Mujoco seems to support headless rendering in a cluster via a handy [script](https://github.com/openai/mujoco-py/blob/master/vendor/Xdummy). I currently use vncserver but sometimes can have unstable connections.

svenevs, 2023-10-22:
@RussTedrake I am having trouble reproducing this.  `bazel test -c dbg //systems/sensors:rgbd_sensor_test` works without issues for the two headless environments I tried:

- SSH'd into a remote machine (that fails to complete any render engine tests for lack of a display server, jammy if relevant).
- Docker running on that same machine (also jammy)

Between when this was filed there was a substantial change in how VTK (the thing needing the display) builds in drake.  Do you still experience this issue on deepnote?  If so, I may need a little help jump-starting how to reproduce the issue.  There are _some_ tricks I have up my sleeve to try and do this on a mac, but it'll require me physically moving one to make it headless (which I can get done later this week), macOS is special in how its display server works so doing from `ssh` while it _has_ a display will not reproduce this either.

Thanks for any tips here!

RussTedrake, 2023-10-23:
I've just tried again [on Deepnote](https://deepnote.com/workspace/russtedrake-b58d8141-6691-43e8-ad76-0575b319c8b8/project/Tutorials-Duplicate-a719f52b-2271-4b35-ace5-06375c07002e/notebook/rendering_multibody_plant-4024bcd2e0154ed1a947a534ec85be27?).  I still get a (jupyter) kernel crash.

My steps to reproduce:
- make a duplicate of the drake tutorials notebook
- Edit the Dockerfile in the project to comment out the DISPLAY env variable:
```
FROM robotlocomotion/drake:jammy-20231016

RUN apt-get -q update && apt-get -q install -y --no-install-recommends nginx-light xvfb && apt-get -q clean

#ENV DISPLAY=:1

ENV PATH="/opt/drake/bin:${PATH}" \
  PYTHONPATH="/opt/drake/lib/python3.10/site-packages:${PYTHONPATH}"
```
- rebuild the docker image if needed (you would be prompted at the top of the Dockerfile display in deepnote)
- Run all cells in the `rendering_multibody_plant` tutorial.  
- scroll down to see 
[image]

jwnimmer-tri, 2023-10-24:
> There are some tricks I have up my sleeve to try and do this on a mac, but it'll require me physically moving one to make it headless (which I can get done later this week), macOS is special in how its display server works so doing from ssh while it has a display will not reproduce this either.

This problem is not related in any way to macOS.  This is an Ubuntu (Deepnote) segfault.

svenevs, 2023-10-24:
Thanks, I will look back into reproducing using Russ's instructions and update here later this week.

svenevs, 2023-10-26:
I have been able to reproduce in a couple of different ways, #20433 should help us get to better error messages rather than a kernel restart (testing is still in progress).

Turns out, it's `SIGABRT` (from `abort()` calls...), not a segfault.

svenevs, 2023-10-27:
I think #20433 is ready for feature review, for whoever would be interested in fulfilling that role :slightly_smiling_face:

jwnimmer-tri, 2023-11-01:
The pull request with a fix is in flight.

It might be worth reinforcing one point here -- one of the challenges here is that **Deepnote does not display stderr to the user**.  Drake _should_ always be using spdlog for messages, not stderr, but that's doesn't always happen when Drake calls into third-party code (e.g., especially with third-party solvers).  It would be extremely helpful if Deepnote could find a way to show stderr to the user, at least in cases where the whole Python interpreter crashed.

closed 2023-11-01 by jwnimmer-tri

jwnimmer-tri, 2023-11-04:
I tested on Deepnote today with 20231104 nightly Docker image.  Looks good:

[image]

If I re-run that cell, the error message gets worse again (Jupyter crashes) but at least the first time we get a helpful error.

PR #20461 (2023-11-01): [render] RenderEngineVtk throws on errors
Previously, it would print to stderr and then abort, which was less-than-helpful for Jupyter sessions.

Closes #18445.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/20461)