RobotLocomotion/drake #18508: Upgrade to Bazel 6.0

jwnimmer-tri, 2023-01-03:
We should upgrade our supported Bazel version to https://github.com/bazelbuild/bazel/releases/tag/6.0.0.

In #18246, I tested a couple of early release candidates and fixed some compatibility problems, so hopefully this is as simple as switching the `setup` scripts (`.bazeliskrc` and `setup/ubuntu/source_distribution`) plus the documentation.

BetsyMcPhail, 2023-01-05:
For reference, #16447 upgraded to 5.0

jwnimmer-tri, 2023-01-05:
FYI one new thing compared to the 5.0 upgrade process --

On macOS today, our bazel version comes from a _dotfile_, which means that CI testing of the PR can use the **Provisioned** images (without any CI maintainer involvement).

The Ubuntu is unchanged -- it comes from a `setup` script, which means that CI testing of the PR needs to use the **Unprovisioned** job names.

rpoyner-tri, 2023-01-18:
Predictably, kcov integration needs love. https://github.com/RobotLocomotion/drake/issues/18615

closed 2023-01-23 by jwnimmer-tri

BetsyMcPhail, 2023-01-23:
Linux provisioned images have been updated in CI.

PR #18545 (2023-01-23): Upgrade Bazel to 6.0.0
This PR is currently in development and serves as a way to test changes on Jenkins during the upgrade to Bazel 6.0.0.

Closes #18508.

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/18545)