RobotLocomotion/drake #18530: CMakeLists Prevents Ubuntu-Derivative OS from Building

cohnt, 2023-01-04:
I'm using the [Tuxedo OS](https://www.tuxedocomputers.com/en/TUXEDO-OS_1.tuxedo) on my computer. It's entirely based on Ubuntu 22.04, and I've been building Drake on it successfully for the past few months. However, the distributor ID when running `lsb_release --id --short` is "Tuxedo", not "Ubuntu" (even though the underlying system really is the same), so compilation instantly fails due to the distribution id check in CMakeLists.txt.

In my case, I can get around this and compile by changing line 56 from `if(NOT UNIX_DISTRIBUTION_ID STREQUAL Ubuntu)` to `if(NOT (UNIX_DISTRIBUTION_ID STREQUAL Ubuntu OR UNIX_DISTRIBUTION_ID STREQUAL Tuxedo))`, but this feels like a bit of a workaround. (And I have to merge the change in every time I update Drake, which is tangling up the commit history of my local repository.)

Would it be possible to enable building on other Linux distributions, along with a warning that such an action isn't supported, instead of failing outright?

Alternatively, it seems to be possible to determine the underlying operating system (regardless of vendor-provided distribution name) from other sources besides `lsb_release`. On my machine, both the `/etc/lsb-release` and `/etc/os-release` files specify the distribution.

`/etc/lsb-release`:
```
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.04.1
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="TUXEDO OS 22.04 LTS"
```

`/etc/os-release` (identical to `/usr/lib/os-release`):
```
NAME="TUXEDO OS"
VERSION="1"
ID=tuxedo
ID_LIKE="ubuntu debian"
PRETTY_NAME="TUXEDO OS 1"
VERSION_ID="22.04"
HOME_URL="https://tuxedocomputers.com/"
SUPPORT_URL="https://support.tuxedocomputers.com/"
BUG_REPORT_URL="https://gitlab.com/tuxedocomputers/development/tuxedo_os/os"
PRIVACY_POLICY_URL="https://www.tuxedocomputers.com/en/Privacy-policy.tuxedo"
VERSION_CODENAME=jammy
UBUNTU_CODENAME=jammy
```

cohnt, 2023-01-04:
You're right! I forgot that I changed that file as well -- I simply added
```
if distro == "Tuxedo":
    distro = "Ubuntu"
```
to that file at line 112. (I'm aware that this is an extremely hacky solution, but it has been working for me so far.)

Not allowing any old Linux distribution makes perfect sense, but enabling Ubuntu derivatives (with an appropriate warning) would be nice. Using the `ID_LIKE` field in `/etc/os-release` seems to be pretty standard practice for these distributions -- it looks like it's also done for PopOS, Elementary OS, and Linux Mint.

jwnimmer-tri, 2023-01-05:
Out of curiosity, could you please share two data points from your Tuxdeo system?

(1) The output of `lsb_release -sc`.  Drake uses both `-si` and `-sc` in various places.

(2) The contents of your `/etc/dpkg/origins/default`.  I think that's where the `Tuxedo` is coming from for `lsb_release -si`.

cohnt, 2023-01-05:
`lsb_release -sc` output: `jammy`
`lsb_release -si` output: `Tuxedo`
`/etc/dpkg/origins/default` contents:
```
Vendor: Ubuntu
Vendor-URL: http://www.ubuntu.com/
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Parent: Debian
```

jwnimmer-tri, 2023-01-27:
This is back on my radar now.  Could I ask for one more data point?  What does `lsb_release -sr` say for your Tuxedo?

Given the above discussion, it seems like it was working for you without any alterations so it is probably `22.04` but I'd like to confirm.

cohnt, 2023-01-27:
Yes, the output of `lsb_release -sr` is `22.04`.

jwnimmer-tri, 2023-01-27:
I have a proposed fix in #18680.  Please let know if it solves the problem in your case.  (You can reply here, or in the pull request.)  Thanks!

cohnt, 2023-04-18:
Sorry to reopen the issue, but it looks like this error pops up in a different way when building with `cmake` instead of `bazel` (which I'm now trying to do -- I should have caught this earlier). When running `cmake ../drake`, I get the following output:
```
CMake Warning at CMakeLists.txt:74 (message):
  Distribution Tuxedo is NOT supported

CMake Error at CMakeLists.txt:656 (file):
  file STRINGS file
  "/home/tommy/opt/rlg/drake/setup/tuxedo/binary_distribution/packages-jammy.txt"
  cannot be read.

-- Configuring incomplete, errors occurred!
```
This is after running `./setup/ubuntu/install_prereqs.sh`, and the correct files are being generated. I'm guessing that `CMakeLists.txt` would need a similar shim as the changes to `tools/workspace/os.bzl` in #18680 to automatically point to Ubuntu instead of the output of the already recorded `UNIX_DISTRIBUTION_ID`. Here's what my setup folder looks like:
```
setup/
...
└── ubuntu
    ├── binary_distribution
    │   ├── install_prereqs.sh
    │   ├── packages-focal.txt
    │   └── packages-jammy.txt
    ├── install_prereqs.sh
    └── source_distribution
        ├── install_prereqs.sh
        ├── install_prereqs_user_environment.sh
        ├── packages-focal-clang.txt
        ├── packages-focal-doc-only.txt
        ├── packages-focal-maintainer-only.txt
        ├── packages-focal-test-only.txt
        ├── packages-focal.txt
        ├── packages-jammy-clang.txt
        ├── packages-jammy-doc-only.txt
        ├── packages-jammy-maintainer-only.txt
        ├── packages-jammy-test-only.txt
        └── packages-jammy.txt
```

I can confirm that renaming the `ubuntu` folder to `tuxedo` solves this issue, as does creating a symlink with `ln -s ubuntu tuxedo`.

jwnimmer-tri, 2023-04-18:
Wow.  I think maybe all CMakeLists lines after 632 might be unused code, with a ticking bomb for Tuxedo.

PR #18680 (2023-02-07): [tools] Improve compatibility with Ubuntu derivatives
Distributions like Tuxedo produce slightly different output from `lsb_release` but actually still work fine.

Closes #18530.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/18680)
PR #19219 (2023-04-20): [cmake] Remove unused CPack code
Closes #18530.

Reverts most of #8188.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19219)
closed 2023-04-20 by jwnimmer-tri