RobotLocomotion/drake #18541: Jammy AMI upgrades (BLAS)

jwnimmer-tri, 2023-01-05:
As of #18540, our Ubuntu 22 Jammy setup dependencies have changed.  Please update the Provisioned Jammy AMIs with the change.  There's no need to change the Focal AMIs.

Note that incrementally re-running the `install_prereqs` script is not enough, since the goal of that PR is to _remove_ packages.  Not only should the openblas packages be removed (`libopenblas-dev libopenblas-pthread-dev libopenblas0 libopenblas0-pthread`), but also `python3-scipy` and any dependencies that it brought in.

It's probable that removing those 5 packages and then `sudo apt autoremove` should do it.  But it's probably worth cross-checking the installed list against an empty machine after `install_prereqs` to see if there's any gotchas.

BetsyMcPhail, 2023-01-05:
It may be easier to start with the Unprovisioned Jammy AMI and run the setup scripts.

jwnimmer-tri, 2023-01-05:
Sure, sounds fine to me.  I wasn't sure how much manual inertia there was in the existing images.  If we can repave from scratch, that'll be the most robust.

BetsyMcPhail, 2023-01-06:
I started with the Jammy unprovisioned image and then ran the [CI setup script](https://github.com/RobotLocomotion/drake-ci/blob/main/setup/ubuntu/install_prereqs) and then the normal [install_prereqs](https://github.com/RobotLocomotion/drake/blob/master/setup/ubuntu/install_prereqs.sh).

Jenkins has been updated and a test run has been started: https://drake-jenkins.csail.mit.edu/view/Linux%20Jammy/job/linux-jammy-gcc-bazel-experimental-release/1732/

BetsyMcPhail, 2023-01-06:
The AMI is still pending... giving it time and will try again.

BetsyMcPhail, 2023-01-06:
Let's try this again: https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-experimental-release/1735/

BetsyMcPhail, 2023-01-06:
Test build passed, closing

closed 2023-01-06 by BetsyMcPhail