RobotLocomotion/drake #18552: Switch documentation build to Jammy

jwnimmer-tri, 2023-01-09:
A sub-task of #18507.

jwnimmer-tri, 2023-01-09:
We have https://packages.ubuntu.com/focal/jekyll at 3.8.6 and https://packages.ubuntu.com/jammy/jekyll at 3.9.0.  As you might imagine, that means diffs in the HTML of the website output (slightly different styles).  I'll need to spot-check that nothing has regressed.

jwnimmer-tri, 2023-01-09:
I've confirmed that when running on Jammy, the documentation seems like it's OK now. 

+@BetsyMcPhail please change the Jenkins jobs [here](https://drake-jenkins.csail.mit.edu/view/Documentation/) to be Jammy-only.

```
linux-jammy-unprovisioned-gcc-bazel-continuous-documentation
linux-jammy-unprovisioned-gcc-bazel-experimental-documentation
linux-jammy-unprovisioned-gcc-bazel-nightly-documentation
```

As part of that, please check that the `drake-ci` scripts for Documentation builds don't hard-code a Focal interlock somewhere, i.e., that the Nightly Documentation will actually publish to the website.

BetsyMcPhail, 2023-01-10:
@mwoehlke-kitware can you please check the `drake-ci` scripts?

BetsyMcPhail, 2023-01-12:
Just to make sure that there's nothing unexpected happening with the CI images, I launched the current Jammy CI job:
https://drake-jenkins.csail.mit.edu/view/Documentation/job/linux-jammy-unprovisioned-gcc-bazel-experimental-documentation/

jwnimmer-tri, 2023-01-12:
FYI @ggould-tri for the v1.12.0 release this is probably relevant to you to deploy the release notes.

ggould-tri, 2023-01-12:
FYI, here are my changes locally that I'm using for my own reference in this release.

[jammy-docs.patch.txt](https://github.com/RobotLocomotion/drake/files/10403877/jammy-docs.patch.txt)

ggould-tri, 2023-01-12:
Documentation build with the new CI for the v1.12.0 release went fine.  Thanks!

closed 2023-01-12 by jwnimmer-tri