RobotLocomotion/drake #18597: Jenkins: Filtered Build Queue is not filtering correctly

BetsyMcPhail, 2023-01-16:
When a tab is selected in Jenkins, e.g. Continuous Production or Experimental, the Filtered Build Queue should also be updated.

svenevs, 2023-01-23:
@jwnimmer-tri this ticket has been repurposed, as far as we could tell there is nothing wrong on the configuration side.  We are optimistic a jenkins update will give us different results, we are somewhat sorely overdue for updating jenkins on whole.

Topic for discussion on Thursday (particularly, when is safe to attempt the upgrade).

jwnimmer-tri, 2023-01-23:
FYI One tip that bit me last week: sometimes, the older versions of plugins fail to classload vs newer versions of Jenkins, to the point where you need to reinstall from backup.  My rule of thumb these days is to upgrade plugins first and then Jenkins second (and wash rinse repeat until no more updates are offered).

svenevs, 2023-10-07:
I was finally successful in getting a vSphere job to launch before restoring the old copy tonight.  We now know how to generate a valid keystore to give to java via the `keytool` command, as well as importing keys from vsphere / github / orka.  We will be sure to document the parts of the upgrade process that mattered internally (right now those notes are incredibly disorganized...  but have all the right information).

We should be able to do this one more time, only we need to

- Retest the strict host key checking being disabled for SSH.
- See what we can do about restoring job logs from the one that's been running most of the time since this maintenance started.

However, we should probably plan to do this over the weekend or on a Friday.  On the backend we have to redeploy the right orka vms.  I'd prefer to just finish it this weekend and let it run for at least one round of nightlies?

jwnimmer-tri, 2023-10-07:
> See what we can do about restoring job logs from the one that's been running most of the time since this maintenance started.

At this point, I think we just give up and tell people their logs might be missing.  That will be OK.

svenevs, 2023-10-09:
Restoring job logs was an easy operation (~20 mins including copy time while doing other things), so the job logs are more or less accurate (possible outliers being anything that happenned during maintenance windows these past few weeks).

Our fully upgraded / updated jenkins appears to be connecting and running with everything.  As far as I can tell all weekly jobs got rerun if they were interrupted, and we're going to see what happens with nightly tonight.  We are able to restore the old one tomorrow morning if something went wrong.

There are followup tasks separated in other tickets, we can close this one this week after a few days of mileage and some backend cleanup (added to checklist on the top of this issue).

BetsyMcPhail, 2023-10-20:
AWS cleanup is complete.

closed 2023-10-20 by BetsyMcPhail