RobotLocomotion/drake #18598: Jenkins: All tabs should be defined in .yaml files

BetsyMcPhail, 2023-01-16:
Currently there is a mix of hand-configured tabs and yaml files.

BetsyMcPhail, 2023-01-26:
@mwoehlke-kitware what's the status with this issue? Do you want to summarize what you discovered and call it closed?

mwoehlke-kitware, 2023-01-26:
The "Wheel" view is now defined in YAML and the description in YAML has been updated to add the drake-external-examples badge that had been manually added to the Production views. The Production views have had their description reset to be an exact match for the other views (just formatting changes, no content changes), for consistency. The Production views regexes have also been updated to be more consistent and simpler / less strict, and are now e.g. `.*-continuous-(?!non-production).*` (before they also had platform restrictions).

The Production views are "dashboard" views, rather than "list" views. This allows them to use "portlets", e.g. the summary of unstable jobs at the top, which is clearly a desired feature. Unfortunately, Jenkins Job Builder only supports "list" views, so the Production views must remain manually-defined.