RobotLocomotion/drake #18602: [py systems] SystemVisitor does not seem to work with diagram recursion?

EricCousineau-TRI, 2023-01-16:
### What happened?

Shown in #18601, as of 3789cd3ef5:
```sh
$ bazel run //bindings/pydrake/systems:py/general_test -- TestGeneral.test_system_visitor
...
Traceback (most recent call last):
  File "{runfiles}/drake/bindings/pydrake/systems/test/general_test.py", line 1034, in test_system_visitor
    system.Accept(v=visitor)
  File "{runfiles}/drake/bindings/pydrake/systems/test/general_test.py", line 1031, in VisitDiagram
    sys.Accept(self)
RuntimeError: Tried to call pure virtual function "SystemVisitor<T>::VisitDiagram"
...
```

### Version

7b17ca24c9398beeae0596939ead4d87e19ba6fa

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

EricCousineau-TRI, 2023-01-16:
Found this when trying to use `SystemVisitor` pattern to recursively collect all systems.

I plan to workaround this simply, e.g.:
```py
def get_all_systems(system):
    systems = []

    def recurse(system):
        if isinstance(system, Diagram):
            for sub_system in system.GetSystems():
                recurse(sub_system)
        else:
            systems.append(system)

    return systems
```

I can revisit after mid-Feb.

@RussTedrake @jwnimmer-tri Any chance y'all have seen this working elsewhere?

jwnimmer-tri, 2023-01-17:
Nope, I've never used SystemVisitor in C++ nor Python.

rpoyner-tri, 2026-11-11:
This looks fixed. `test_system_visitor` in `bindings/pydrake/systems/test/general_test.py` on master is the scenario from this issue: a Python `SystemVisitor` whose `VisitDiagram` calls `sys.Accept(self)` on each subsystem. I also ran an equivalent visitor with the Drake 1.58.0 release wheel (Linux, Python 3.12) and it recursed without the "pure virtual function" error. I didn't track down the specific commit that fixed it.

Closing as completed; please reopen if you still see it.

closed 2026-11-11 by rpoyner-tri