RobotLocomotion/drake #18619: Release v1.12.0 docker and apt

BetsyMcPhail, 2023-01-18:
From #18556

 https://github.com/RobotLocomotion/drake/releases/tag/v1.12.0

The release date is 2023-01-12
The SHA is https://github.com/RobotLocomotion/drake/commit/643caa184dc645cf47775ed75d7a2052ab723dc7
The release tag is v1.12.0

BetsyMcPhail, 2023-01-18:
@mwoehlke-kitware is going to follow the newly created release documentation to make sure that more than one person has tested the process.

BetsyMcPhail, 2023-01-18:
Matthew noticed that: [https://github.com/RobotLocomotion/drake/releases/tag/v1.12.0](https://github.com/RobotLocomotion/drake/releases/tag/v1.12.0) does not match [S3](https://s3.console.aws.amazon.com/s3/buckets/drake-packages?prefix=drake%2Frelease%2F&region=us-east-1&showversions=false# )-- we're skipping the arm64 releases.

@svenevs noted that [v1.8.0](https://github.com/RobotLocomotion/drake/releases/tag/v1.8.0) is the first non-arm64 enabled build so we probably should manually update the s3 releases bucket to add the arm64 tarballs? (In addition to making sure future ones get uploaded too)

jwnimmer-tri, 2023-01-18:
Could you give one specific example of the discrepancy (e.g., for v1.11.0 which exact filenames are missing or extra, at which locations)?

My mental model is that the GitHub release page is the source of truth for our binaries of the numbered releases, and that `s3://drake-packages/drake/release/` is merely a backup in case GitHub gets nuked.  The S3 releases are not documented anywhere that users could find or use, not even the `https://drake-packages.csail.mit.edu/drake/release/` is documented.  Users should always use the GitHub download page.

Under that model, the script(s) that sync from the GitHub release page into S3 should not contain hard-coded any lists about what to sync -- all attachments on GitHub should be synced, nothing more nothing less.  (In particular, no new checksum files nor signatures.)  That would mean that if we've failed to sync attachments to S3 for prior releases, we should do that in order to finish the "backup" procedure.

Notably this is only for stable releases.  The ephemeral nightly & experimental binaries do live on S3 under drake/nightly and that's fine.  No GitHub involved.

Also distinct is the aptly site.  The push_release stuff does need to take the deb from GitHub (or, I suppose, our backup of GitHub) and copy/reindex that into the aptly bucket.

Hopefully that all makes sense?

BetsyMcPhail, 2023-01-19:
For example, both 'mac' filenames exist:

https://github.com/RobotLocomotion/drake/releases/download/v1.11.0/drake-20221214-mac.tar.gz
https://s3.console.aws.amazon.com/s3/object/drake-packages?region=us-east-1&prefix=drake/release/drake-1.11.0-mac.tar.gz

Whereas the 'arm64' version is only on Github:

https://github.com/RobotLocomotion/drake/releases/download/v1.11.0/drake-20221214-mac-arm64.tar.gz
https://s3.console.aws.amazon.com/s3/object/drake-packages?region=us-east-1&prefix=drake/release/drake-1.11.0-arm64.tar.gz

It seems like noticing this discrepancy uncovered the root of the issue, that the [push_release](https://github.com/RobotLocomotion/drake/blob/master/tools/release_engineering/dev/push_release) script does not directly copy from the Github release page to S3.  Instead it:

* Downloads the nightly binary release archives for focal, jammy, and mac from S3 and renames them with the x.y.z version number.
* Downloads the source release archive from GitHub (e.g. https://github.com/RobotLocomotion/drake/archive/v1.11.0.tar.gz) and renames it with the x.y.z version number and the suffix src (e.g. `drake-1.11.0-src.tar.gz`)
* Generates detached armored GPG signatures and SHA256 and SHA512 checksums of the renamed binary and source release archives
* Uploads each archive, signature, and checksum to S3.

Unfortunately, this script wasn't under source control until recently so I don't have any version history. I'm guessing some of it was remnants of the old apt process?

jwnimmer-tri, 2023-01-19:
Summary of f2f: the most important parts for 1.12 release are to get the docker and aptly pieces deployed.  We should also do the S3 archiving in line with what we've been doing in the past (even though it's odd/incomplete).

In the future, we'll rewrite the tool in Python and change it to work the way we want to, even to the point of having Jenkins invoke the tool instead of us manually.  We should open a new issue for that task, though.  This one can stay scoped to just 1.12 administrivia.

BetsyMcPhail, 2023-01-19:
The existing release procedure has been followed for docker / apt / S3.

Matthew has a successful environment set-up for everything except the final apt step and Stephen has experience with the apt packaging, so we have redundancy on the whole process, just in case.

Calling this issue closed.

closed 2023-01-19 by BetsyMcPhail