RobotLocomotion/drake #18638: Externals upgrade cohorts

jwnimmer-tri, 2023-01-23:
For upgrading our externals, we have a some institutional knowledge about externals that should upgrade in groups.

For example `usockets` + `uwebsockets` are highly coupled, so when upgrading them we should use a single commit with both bumps, instead of two separate commits.

Ditto for `sdformat_internal` + `gz_math_internal` + `gz_utils_internal`.

I think it'd make sense to teach `new_release.py` about this, e.g., with a constant atop the file:

```py
# Packages in these cohorts should be upgraded together (in a single commit).
_COHORTS = (
    # The uwebsockets depends on usockets; be sure to keep them in aligned.
    (`uwebsockets`, `usockets`),
    # The sdformat depends on both gz libraries; be sure to keep them in aligned.
    (`sdformat_internal`, `gz_math_internal`, `gz_utils_internal`),
)
```

This automation should help keep the process simpler for the developers, and reduce glitches with unsync'd versions.

BetsyMcPhail, 2023-01-23:
Would it make sense to also add a note to the appropriate `repository.bzl` files? They often contain a comment with updating hints or instructions.

jwnimmer-tri, 2023-01-23:
Yes, probably so.  My primary aim is to automate it to reduce mistakes, but providing extra context in comments would help future readers understand.  (Possibly the `repository.bzl` comment should cite the cohorts scripting, so that there is only one source of truth for what is and isn't a cohort.)

BetsyMcPhail, 2023-02-07:
Per f2f, @mwoehlke-kitware is going to finish up this issue and try the new script on this month's updates.

mwoehlke-kitware, 2023-02-07:
Looking at this month's upgrades, I notice mypy_extensions_internal and mypy_internal. I'm not *certain* these are co-dependent, but probably makes sense to make them a cohort anyway?

jwnimmer-tri, 2023-02-07:
Yup, SGTM.

(The overall goal is to help automate and/or reduce mistakes, so really anything you yourself observe during any monthly upgrades is probably fair game.)