RobotLocomotion/drake #18639: ModelVisualizer slow loading time

jwnimmer-tri, 2023-01-23:
In the case the model being visualized is very complicated (e.g., Ironwood in Anzu), it might take 5-10 seconds to fully load.  Meanwhile, the command line happily says "Use Esc to Exit" as if the model is ready to go, but the window shows an empty scene.

The slow loading for Meshcat overall is tracked elsewhere, but the UX for ModelVisualizer can do better.

I think we need a progress message, like "Loading..." while it's loading, and then something like "Ready" (or the current "Esc" message) once it's loaded.

In the code, we can call `meshcat.Flush` to block until the model is fully loaded.

trowell-tri, 2023-02-20:
The simplest thing that seems like it should work is:

```
--- a/bindings/pydrake/visualization/_model_visualizer.py
+++ b/bindings/pydrake/visualization/_model_visualizer.py
@@ -300,6 +300,10 @@ class ModelVisualizer:
             self._plant_context,
             self._meshcat)])
 
+        print("Loading...", end="")
+        self._meshcat.Flush()
+        print(" done.")
+
         # Wait for the user to cancel us.
         stop_button_name = "Stop Running"
         if not loop_once:
```

I wasn't able to find a model that didn't load immediately for me until I tried a [dozen IIWA arms](https://tri-internal.slack.com/archives/C02KXJ6LM54/p1676509363773709?thread_ts=1676477456.086119&cid=C02KXJ6LM54); that "model" does take a while to visually load but there's no detectable pause at the `meshcat.Flush()` call. I read through the `Flush` code and parts of the various Publish methods but wasn't enlightened, and the existing code I checked out didn't indicate that I'm using this API wrong…

jwnimmer-tri, 2023-02-20:
The question would be if all of the messages are queued up by the time we hit the Flush().  If we're sending big messages _after_ Flush, then the Flush wouldn't help enough.  However, by my reading the `ModelVisualizer.Finalize()` function should be pushing all of the big meshes already, since it calls `self._diagram.ForcedPublish(self._context)`.

Thinking more, the problem maybe is that the browser isn't even there yet.  It's not until _after_ a browser connects that we send it all of the data.  A call to `meshcat.Flush()` with no browsers connected is a no-op.

A more sensible plan might be:
- Print a message like "Open your browser to the URL".
- In case `meshcat.GetNumConections()` is zero, spin until we see it reach >0, then print "Loading...", then Flush, then print "Ready.

We'd want to be careful that anyone using `meldis` (only) to preview can still do so.  I don't think it requires any special handling.

trowell-tri, 2023-02-21:
I was pretty sure that `ForcePublish` was where the objects would get sent, but before I began diving into the publish code before I had tried calling `Flush` in other places and never found that it blocked.

If you pass the "open window" flag to model visualizer, `meshcat.GetNumActiveConnections()` returns `1` before the `Run` method even starts and never increments again; if you don't pass that flag and await that number incrementing as you open the URL, `Flush` still fails to block and you can see models still loading after the "Click Stop Running" message:

```
--- a/bindings/pydrake/visualization/_model_visualizer.py
+++ b/bindings/pydrake/visualization/_model_visualizer.py
@@ -300,6 +300,15 @@ class ModelVisualizer:
             self._plant_context,
             self._meshcat)])
 
+        if self._meshcat.GetNumActiveConnections() == 0:
+            print("Waiting for browser")
+            while self._meshcat.GetNumActiveConnections() == 0:
+                time.sleep(1 / 32.0)
+
+            print("Loading...", end="")
+            self._meshcat.Flush()
+            print(" done.")
+
         # Wait for the user to cancel us.
         stop_button_name = "Stop Running"
         if not loop_once:
```

One interesting thing I noticed from running this over and over is that when passing the "new window" flag, the meshcat window  initially shows what I now assume is every sub-object (i.e. every individual link in each arm) being loaded in at the origin of the world and only then moving to their final positions (the last part happens pretty fast so I may be misinterpreting and I'm only seeing one arm being loaded, but I think I see pieces disappearing). But neither phase seems to block `Flush` … is the latency *not* from sending data?

jwnimmer-tri, 2023-02-21:
I tried this patch as well, in case a round-trip to the javascript would help:

```
--- a/bindings/pydrake/visualization/_model_visualizer.py
+++ b/bindings/pydrake/visualization/_model_visualizer.py
@@ -303,6 +303,13 @@ class ModelVisualizer:
         # Wait for the user to cancel us.
         stop_button_name = "Stop Running"
         if not loop_once:
+            print("Loading...")
+            import time
+            time.sleep(3)
+            self._meshcat.Flush()
+            self._meshcat.AddButton("Loading...")
+            self._meshcat.GetButtonClicks("Loading...")
+            self._meshcat.DeleteButton("Loading...")
             print(f"Click '{stop_button_name}' or press Esc to quit")
 
         loop_result = RunResult.KEEP_GOING
```

Still no dice.

I don't think there's any easy way to ask Meshcat whether the scene is finished displaying in the browser.  So, this feature isn't worth any more time.  Closing as WONTFIX.

trowell-tri, 2023-02-21:
I guess this makes me ask: when does Flush block? From reading Slack history it seems it had some effect…

closed 2023-02-23 by jwnimmer-tri