RobotLocomotion/drake #18640: Reorg of planning-related code

jwnimmer-tri, 2023-01-23:
As part of making our motion planning toolbox easier for everyone to understand, we are going to rework some directory/module names of the existing features, so it's easier for people to discover and browse through our tools.

Specifically, to relocate:

- [ ] From `manipulation/planner:constraint_relaxing_ik` into `multibody/inverse_kinematics:___` (i.e., with the same filename, just a new directory name).
- [ ] From `manipulation/planner:differential_inverse_kinematics` into `multibody/inverse_kinematics:___` (likewise).
- [ ] From `manipulation/planner:differential_inverse_kinematics_integrator` into `multibody/inverse_kinematics:___` (likewise).
- [ ] From `manipulation/planner:robot_plan_interpolator` into `manipulation/util:robot_plan_interpolator`.
- [ ] From `systems/trajectory_optimization:*` into `planning/trajectory_optimization:*` (all files).

For any of these that are Stable API, we'd introduce deprecation shims to that the old include paths and namespaces continue to work during a transition period.

We'd like to enact this change prior to the next release in mid-February.

\CC @RussTedrake @calderpg-tri @SeanCurtis-TRI FYI

RussTedrake, 2023-01-24:
FWIW -- The systems/trajectory_optimization is the only one that will be disruptive to my spring class, and trajopt is a few weeks out yet: [schedule](https://underactuated.csail.mit.edu/Spring2023/schedule.html).

SeanCurtis-TRI, 2023-01-24:
@RussTedrake quick clarification question: does your class use master or a pegged release? I.e., if I crank on this now and get this resolved in the next week, is that the best outcome for your class?

RussTedrake, 2023-01-24:
I use docker so can use nightlies or proper releases, but prefer to use releases when possible.  Getting it resolved in the next week would be great.  Thanks!

jwnimmer-tri, 2023-01-31:
@RussTedrake just to double-check in case you have any objections, for the trajectory optimization Python bindings our plan is to use the `pydrake.planning` module, as in ...

```py
# This is correct import path.
from pydrake.planning import DirectCollocation
```

... versus ...

```py
# This won't work: no such module "trajectory_optimization".
from pydrake.planning.trajectory_optimization import DirectCollocation
```

... along the lines of #11802.

RussTedrake, 2023-02-01:
SGTM

SeanCurtis-TRI, 2023-02-13:
The final task is to revisit the current [https://drake.mit.edu/doxygen_cxx/group__planning.html](https://drake.mit.edu/doxygen_cxx/group__planning.html) with more info. Given that I'm not sure what information qualifies as helpfully "more", I suspect someone closer to this planning code should probably address this. The linked google doc doesn't shed particular light onto the topic. The only inference I can safely make is to pull `ConstraintRelaxingIk` out of the group so that it isn't advertised to users as the first in the list -- as it has been described as "not really great".

If anyone has some insight to share, I'm happy to tackle this. If, however, we can't articulate what issues are being addressed with the documentation, I propose we close this issue.

jwnimmer-tri, 2023-02-15:
My thought on https://drake.mit.edu/doxygen_cxx/group__planning.html was relatively simple -- make it an appropriate table of contents for what we have, with nothing left out.  (For example, I think CollisionChecker is not listed currently.)  Broadly, anything the planning folder should probably be reachable from this ToC, either directly or via some intermediate sub-pages.

Future work would be adding more introductory material and/or tutorials, but that isn't part of this ticket.  Here, I just want to ensure that all of the new code is appropriately indexed onto the overview page.

PR #18854 (2023-02-23): [planning] Update algorithms/planning module documentation
1. Adds the new classes in drake/planning to the module.
2. Adds subgroups to the module to provide broad guidance on the classes.
3. Moves the planning module documentation into its own header.

resolves #18640

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/18854)
closed 2023-02-23 by EricCousineau-TRI