RobotLocomotion/drake #18656: [pybind] Keep alive, refererence (tr.) usages and semnatics are confusing / possibly incorrect

EricCousineau-TRI, 2023-01-26:
### What happened?

See discussion in https://github.com/RobotLocomotion/drake/pull/18650

Given that we haven't seen acute errors, beyond things like #14387 that can be worked around, I won't plan to tackle it anytime soon. However, good to note!

\cc @SeanCurtis-TRI @jwnimmer-tri

jwnimmer-tri, 2023-01-26:
Mostly my disposition here overall is that we should switch to `shared_ptr` for our Python/C++ interfacing so that everything becomes a lot more obvious and natural, ala #13058 and related.

rpoyner-tri, 2025-03-19:
Since this was filed, the RobotDiagrams bindings have been completely re-annotated with custom techniques, and pretty much all of the unique_ptr stuff has been evicted or quarantined. I'm sure there are still keep_alives that are either immortality hazards (#22515) or flat wrong, but specific complaints would serve better at this point. Closing.