RobotLocomotion/drake #18659: Rewrite/refesh of push_release tooling

jwnimmer-tri, 2023-01-26:
See https://github.com/RobotLocomotion/drake/issues/18619#issuecomment-1396216666 for some background and details.

There's enough going on there that we think rewriting in Python will probably be to our advantage.

The basic requirements are:

(1) Copy all attachments from the GitHub release page into S3 for backup purposes (in case GitHub loses them somehow).

(2) Inject the `*.deb` files downloaded from the GitHub release page into the aptly site.

(3) Push the version-numbered docker tags.

At the time of the rewrite, we should support support macOS for this tool; it should target Ubuntu Jammy (only).

---

Also a note in case it helps for planning -- for (3) at the moment we just re-tag the nightly docker images as stable.  In the future, the `*.tar.gz` files associated with a stable release will embed a stable version number, not a nightly version stamp, at which point we'll want to create new docker images using those tarballs, rather than re-tagging nightly tag.  If it makes sense, the new script could work that way from the get-go (mimicing [drake-ci](step-build-docker-image.cmake), where it takes the downloaded `*.tar.gz` file from GitHub, unpacks it into a local Docker, runs `install_prereqs`, and drops a tag.

jwnimmer-tri, 2023-05-24:
Addendum for clarity: currently, the script is GPG-signing (`*.asc`) the https://drake.mit.edu/from_binary.html tarballs.  It should stop doing that.  We don't advertise that signature in our docs, and it introduces needless complexity.  Only the requirements listed in the top post should remain intact in the rewrite.

BTW related: https://blog.pypi.org/posts/2023-05-23-removing-pgp/

BetsyMcPhail, 2023-05-31:
When we eventually get to implementing a job in Jenkins, it is possible to "Enable project-based security" and create a security matrix that does not inherit from the global security settings. This way, we can specify users or, more likely, a Github group that is allowed access to the job.

jwnimmer-tri, 2023-06-04:
For some updates, see https://github.com/RobotLocomotion/drake/pull/19496#pullrequestreview-1444773578.

> This means that the *.deb.asc file needs to appear on the GitHub releases page one way or another.

Note that (4) is gone now.  We will no longer be signing individual release files, rather only the aptly Releases index.

jwnimmer-tri, 2023-06-29:
With the new scripts, we've improved and automated this a lot more.  There's more we could do, but we might as well defer that until we switch more automation over to Jenkins instead of manual scripts.

TODO(Jeremy) -- Open issues for the follow-up:
- push the source tarball as a GH release attachment.
- use better deb names in the GH release attachment.

jwnimmer-tri, 2023-07-06:
> push the source tarball as a GH release attachment.

https://github.com/RobotLocomotion/drake/issues/19734

> use better deb names in the GH release attachment.

https://github.com/RobotLocomotion/drake/issues/19735